iNavFlight/inav #1857: Altitude calculation and descending problem

pstryju, 2017-07-02:
Hi. I've got a problem with altitude. Not exactly with hold, because when it's not changing quad remains stable, also there is no problem when it's going up, only when going down. Quad just don't want to lose altitude even with throttle on 0. Or it's going down for some meters, stops and again.
Also I've noticed weird altitude calculation visible on sensors tab.

[image]

I've moved quad about 80cm up and stopped. At the beginning altitude is calculated properly, even baro behaves good, but Alt is going back to even smaller level than was before moving.
I think that acc calculation is proper because fast changes are good, later it drifts despite not drifting baro sensor.

Not attaching settings dump, because it's the same even after reflashing board so settings are default.

Here is blackbox dump when landing during RTH failed (last flight).

[blackbox_log_2017-07-02_215529.TXT](https://github.com/iNavFlight/inav/files/1118103/blackbox_log_2017-07-02_215529.TXT)

F450, 920kv, 30A SimonK, Naze32 Rev6 Full, NZ GPS Ublox-7, 1045 props

Ralf-W, 2017-07-03:
From the BB-Log your acc(z) values are jumping much too strongly - probably due to vibration problems - at high throttle settings. Since iNAV is calculating altitude by integrating acc(z) over time this results in wrong altitude data.

DzikuVx, 2017-07-04:
I agree here. To much vibration is causing this kind of problems. Balance propellers and motors, check if nothing is loose or broken

digitalentity, 2017-07-04:
Yes, most likely vibration issues - accelerometer is very jumpy.

DzikuVx, 2017-07-04:
@pstryju I'm closing this ticket since it appears that vibrations are your problem. If any new data appears, feel free to reopen

closed 2017-07-04 by DzikuVx