iNavFlight/inav #1860: Problem OSD and Smartport at Softserial on a SPRacingF3

DerDominik, 2017-07-04:
Hi there,

I want to use a MicrominimOSD and the Smartport telemetry at both softserial ports on a SPRacingF3, but I got always "No Data" message on my OSD while both are activated. If a disable telemetry, the OSD is working fine.
Is this a problem with INav and/or the FC? Or is this combination generally not working?

DzikuVx, 2017-07-05:
I can confirm that MSP and SmartPort do not work together on SoftSerials here. 
You might want to switch MinimOSD to LTM protocol, that might help.

closed 2017-07-05 by DzikuVx

DerDominik, 2017-07-05:
Thanks for your help, I'll try later.
Only change the setting in the MicrominimOSD?
Or do I've to change something else in iNav?

DzikuVx, 2017-07-05:
You have to configure MW OSD to use LTM and set LTM as telemetry on that SoftSerial