iNavFlight/inav #1861: HMC5883 MAG=UNAVAILABLE on Flip32AIO flashed with custom inav 1.7.1

georgekucher, 2017-07-04:
Hello guys!
I'm trying to connect my external HMC5883 compass to FC through I2C bus for a week, but no luck(((
Here is some output:

>## status
>System Uptime: 1348 seconds
>Voltage: 1 * 0.1V (1S battery - NOT PRESENT)
>CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611
>Sensor status: GYRO=OK, ACC=OK, MAG=UNAVAILABLE, BARO=OK, SONAR=NONE, GPS=NONE
>Stack size: 2048, Stack address: 0x20005000
>I2C Errors: 0, config size: 1674, max available config: 2048
>System load: 4, cycle time: 2004, PID rate: 499, RX rate: 49, System rate: 9

>## dump
>### INAV/NAZE 1.7.1 Jun 12 2017 / 19:32:46 (48e1feff)
>mixer QUADX
>
>mmix reset
>
>servo 0 1000 2000 1500 100 -1 
>servo 1 1000 2000 1500 100 -1 
>servo 2 1000 2000 1500 100 -1 
>servo 3 1000 2000 1500 100 -1 
>servo 4 1000 2000 1500 100 -1 
>servo 5 1000 2000 1500 100 -1 
>servo 6 1000 2000 1500 100 -1 
>servo 7 1000 2000 1500 100 -1 
>smix reset
>
>feature -RX_PPM
>feature -VBAT
>feature -RX_SERIAL
>feature -MOTOR_STOP
>feature -SERVO_TILT
>feature -SOFTSERIAL
>feature -GPS
>feature -TELEMETRY
>feature -CURRENT_METER
>feature -3D
>feature -RX_PARALLEL_PWM
>feature -RX_MSP
>feature -RSSI_ADC
>feature -LED_STRIP
>feature -DASHBOARD
>feature -BLACKBOX
>feature -CHANNEL_FORWARDING
>feature -TRANSPONDER
>feature -AIRMODE
>feature -SUPEREXPO
>feature -VTX
>feature -RX_SPI
>feature -SOFTSPI
>feature -PWM_SERVO_DRIVER
>feature -PWM_OUTPUT_ENABLE
>feature -OSD
>feature RX_PPM
>feature VBAT
>beeper GYRO_CALIBRATED
>beeper HW_FAILURE
>beeper RX_LOST
>beeper RX_LOST_LANDING
>beeper DISARMING
>beeper ARMING
>beeper ARMING_GPS_FIX
>beeper BAT_CRIT_LOW
>beeper BAT_LOW
>beeper GPS_STATUS
>beeper RX_SET
>beeper ACTION_SUCCESS
>beeper ACTION_FAIL
[... 364 lines cut ...]
>set nav_mc_vel_z_i = 50
>set nav_mc_vel_z_d = 10
>set nav_mc_pos_xy_p = 65
>set nav_mc_pos_xy_i = 120
>set nav_mc_pos_xy_d = 10
>set nav_mc_vel_xy_p = 180
>set nav_mc_vel_xy_i = 15
>set nav_mc_vel_xy_d = 100
>set nav_fw_pos_z_p = 50
>set nav_fw_pos_z_i = 0
>set nav_fw_pos_z_d = 0
>set nav_fw_pos_xy_p = 75
>set nav_fw_pos_xy_i = 5
>set nav_fw_pos_xy_d = 8
>set rc_expo = 70
>set rc_yaw_expo = 20
>set thr_mid = 50
>set thr_expo = 0
>set roll_rate = 20
>set pitch_rate = 20
>set yaw_rate = 20
>set tpa_rate = 0
>set tpa_breakpoint = 1500

Any help is very needed

Ralf-W, 2017-07-04:
Power from USB cable is not sufficient to drive external MAG (+GPS). Did you connect an external 5V supply?
Did you try to exchange Rx and Tx cables from MAG on I2C connector?
Was internal MAG detected correctly before you disconnected it mechanically?

Schalonsus, 2017-07-05:
Is it 3.3V or 5V Mag board?

DzikuVx, 2017-07-05:
I also suspect it might be USB power connected. Onboard voltage stabilizer might be not strong enough, or even not powered when working only on USB.

georgekucher, 2017-07-05:
First of all I have no internal MAG, that's why I bought this board
http://www.robotpark.com/image/data/PRO/91457/GY_271_ELECTRONIC_COMPASS.pdf
As you can see, it is 3-5V.
Today I've tried to connect 4S lipo to my FC, but no luck.
Also I tried supplying MAG with 3.3V and 5V.
Connect external 5V (Power Bank) to FC and to MAG (when it was connected to 5V supply)
Voltage is present on MAG in every case. But it is still UNAVAILABLE in configurator.
And of course I've tried to swap SDA SCL lines, no luck.
Maybe you know something about "i2c_overclock" setting? Should it be on or off?
P.S. I had familiar troubles with connecting my BT-GPS module to FC. But as I found out it was a soft problem. I rebuilt custom FW (with NMEA protocol) and it worked. But here - I'm out of ideas. Wish I had oscilloscope, but now it's only multimeter((

Thanks a lot for all your answers. Maybe there is something I missed?...

giacomo892, 2017-07-05:
Check if SDA SCL has some voltage when you power the module up (disconnected from the FC)

georgekucher, 2017-07-05:
Yes it has almost the same voltage as its power supply VCC (3.3V if supplied 3.3V and 5V i suppled 5V)

giacomo892, 2017-07-05:
I you have an arduino lying around try to connect the mag to it and check if it is working :)

georgekucher, 2017-07-05:
Sadly I have no arduino((
But maybe there is another way to check it? I have pl2303 board. And some other old hardware...

georgekucher, 2017-07-05:
By the way. I read about some kind of pullups (4.7K resistors between SDA - VCC and SCL - VCC). May it help?

Allerhopp, 2017-07-05:
Usually there are internal 1k pullups, you can measure them with a multimeter between SDA - VCC and SCL - VCC. If there are no internal pullups it is surely worth a try to add them.

georgekucher, 2017-07-05:
there are 2.18kOhm & 2.18kOhm between them

Allerhopp, 2017-07-06:
OK. In my pov. it does not matter, where the pullups are. Presumably there are also pullups in your Flip32. Out of interest, can you measure the Flip pullups also? If there are also pullups and you can remove the MAG pullups easily ......

georgekucher, 2017-07-06:
But I didn't put there any pullups...

Allerhopp, 2017-07-06:
Sorry, for not being clear. My suggestion is, if there are pullups on both sides, to remove them on the mag, if this is possible. Then try again, if the mag is detected by the Flip. But i see only a little chance, this will do the job.

KiwiDavid, 2017-07-07:
Hi,
The HMC5883L is rated 3.6V max and is typically powered off a 3.3V regulator associated with the GPS chip.  5V on its supply is in excess of it's absolute max rating.   The GPS will require an external battery to be powered up and I wouldn't expect the USB to run GPS and mag.   If the GPS is active then so should the mag sensor be working as it draws only very small currents.

The pull-up resistors are not critical and are usually located nearer the host rather than the slave. The I2C bus is active low but passive (resistive) high. Provided the resistors are low enough in value to provide a fast +ve going edge for clock and data that is all that matters.  The open drain outputs provide the active low signal.  The resistors are driving the total bus capacitance  - the more slave devices and bigger the layout the lower the resistors have to be in value.   I doubt you would need to go under 2k2 even at 400kHz clock rate.

I'm having problems with my external mag sensor (on SPRacing F3) not being recognized but mine is temperature sensitive - it won't initialize under about 20C but is fine above that. I have a scope and logic analyzer and I can see that below 20C the host micro never addresses the mag sensor.
If you make a small change in an unrelated setting and do a Save/Reboot does the mag sensor suddenly appear?  (make sure the GPS is powered first.

Cheers,
David

KiwiDavid, 2017-07-07:
Hi again,
Sorry - I may have scared you - the mag device you have has a 5V to 3.3V regulator on the board so you can feed the supply pin with 5V but not directly on to the mag sensor chip itself.
A word of caution - your circuit also has 3.3V to 5V non-inverting level translating.   The host micro won't really like having 5V logic sent to it.   The pull-ups are the 4k7 resistors but you may be better coming directly off the mag sensor and remove the two discrete logic translating FETs and the 4 resistors.  This would make it basically the same as GPS/ mag sensor combo.
I don't think you will have damaged anything because the currents would be limited but I think your micro would be expecting 3,3V logic rather than 5V.
Cheers,
David

georgekucher, 2017-07-08:
Hello!
I was at my friend's (he has an oscilloscope) and we measured signals at SDA, SCL pins at FC and MAG separately. So FC sends signals 500Hz (he might be wrong in calculations because his oscilloscope is analog and very old) but MAG sends nothing (straight line which is on VCC level 3.3V on both SDA and SCL pins). Does it mean that MAG is dead, or it is just waiting for some signal from FC to start work??
Also I measured pullups on FC, they are 1k both, so when I plug MAG, overall impedance is near 0k67 on each (SDA-VCC and SCL-VCC).

KiwiDavid, 2017-07-08:
Hi,
The I2C bus is common to the FC, gryo/acc, mag and baro (if fitted) so looking at the SDA or SCL on any device should see the total bus activity.  If you don't see activity on the mag SDA and SCL lines then it's not connected.
The FC is master and will send commands so even if the mag sensor was dead you would see activity unless the both lines were shorted to 3v3 but you have already confirmed you see bus signals elsewhere.
Are you sure you have connected to the correct place?
You still have the issue of the level translation being 5V outputs and not 3v3.  Your mag widget is trying to pull the bus lines to 5V.  I would at least remove the resistors from drain to 5V on your board and let the translation FETs share the FC's pullups but ideally you should remove all the level translation circuit.
Good luck.

Cheers,
David

georgekucher, 2017-07-09:
Hello,
As for wiring, I connected according to [this diagram](https://www.google.com.ua/search?q=flip32+pinout&source=lnms&tbm=isch&sa=X&ved=0ahUKEwiDsKGc69bUAhVjJJoKHbb2AbEQ_AUICigB&biw=1366&bih=662#imgrc=UXp-DfYBBL_cyM:).
Answering to your previous questions:
It not appears after I make small change in config and save/reboot, even if it is ON pretty long. By the way, my GPS and MAG are the separate boards.
And about 3.3V and 5V logic:
Does my MAG board try to pull the logic levels to 5V even if it is powered with 3.3V? If so, could you please mark the components which should be removed, sadly I'm not very good electrician(( Here is the photo: [GY-271](https://www.google.com.ua/search?q=gy-271+datasheet&source=lnms&tbm=isch&sa=X&ved=0ahUKEwi_oObS0PrUAhVqApoKHcjnCDsQ_AUICigB&biw=1366&bih=662#tbm=isch&q=gy-271&imgrc=5B8ydyY0kUtgFM:)

KiwiDavid, 2017-07-09:
Hi,
Well it's clearly marked for SDA and SCL so provided you have the correct connector and it's wired from SDA on the FC to SDA on the mag sensor and similarly for SCL you should be ok.  Have you also selected the HMC5883 in the sensor list under the configuration tab?

Your mag board has level translation FETs to change the sensor's 3.3V output back up to 5V.  These resistors will be trying to pull the whole I2C bus up to 5V but I suspect it will only reach about 3.9V when some of the input protection diodes on the other sensors and the micro begin to conduct and clamp the voltage.
Ideally you need to remove the 4 parts marked 222 (2k2 resistors) and the 6pin IC (2x FETs).
You then need to run a small wire (fine enamel wire) from pin 1 of the HMC5883 directly to the SCL output pin and another from pin 16 of the HMC5883 to the SDA output pin.   Rather than try to solder to the sensor pins you can follow the track to the resistors or the pads of the 6 pin IC and solder from there.
This will then leave you with a mag sensor powered off a 5V line but regulated down to 3.3V.  It will have no pull-up resistors and hence will share those on the FC itself.  Output level will be 3.3V as dictated by the FC's power supply.
If your GPS is working don't change anything.  Just make sure you have selected the mag sensor as mentioned above.  Try this first if you haven't already - you may find it works without modifying the level translation circuitry.
Good luck.

Cheers,
David

georgekucher, 2017-07-09:
Thanks for your answers, David.
Yes I'm connecting exactly as you describe and setting HMC5883 in configuration from the start.
And thanks for the detailed description of what I have to do. I will try and let you know.
Have a nice day!

georgekucher, 2017-07-09:
I did everything like you said and still nothing((((
The strange thing is that during all my attempts I never saw I2C errors, it's always zero

KiwiDavid, 2017-07-10:
Hi George,
It still seems like the SDA and SCL lines are not connecting.  They should idle high at 3.3V and pulse low with I2C bus activity.  On a busy line they will be low almost as much as they are high so if reading the average voltage they would between about 1.6V and the 3.3V idle state.
Is there any chance the mag module is not being earthed back to the same ground as the micro?  That would cause some strange errors especially if measuring voltage at the mag sensor.  Are the pins in the connector still there and not bent of broken off?
You mentioned earlier your friend had seen bus activity but not on the mag module - this indicated that the module wasn't connected properly.
You may be able to use a multimeter to check for continuity between the back of the FC's I2C port connector and the respective SDA and SCL on the mag module.  Also check the ground for continuity. Do this with no power connected.
Please get back if you have any further information.

Cheers,
David

georgekucher, 2017-07-12:
Hi David,
All connections are good. I even caught couple of I2C errors when connect SDA - SCL (or swapping them) while FC is powered ON, which means that pins on FC are correct (I2C). But no MAG was detected (still MAG=UNAVAILABLE). I've ordered an Arduino UNO and gonna to check the MAG with it. Is there any known models of external MAGs which are 100% working with my Flip32 FC (not very expensive).

KiwiDavid, 2017-07-12:
Hi George,
This is not good.  I can't imagine that the FC has anything to do with it.  The sensor is stand-alone with no other circuitry associated with it and even the pull-up resistors are shared with the other sensors so they are in place and working.
I checked the I2C bus speed on my SP Racing F3 iNav install (1.7.1) and it's running 400kHz.  This is the top speed for the sensor but still within its spec.  I don't know much about the Flip32 but I2C is I2C so if it's reading gryo/acc and baro (?) the mag sensor should be no problem.
It may pay to ask the question on the general forum for recommended mag sensors but many people use the GPS/mag combo.   This is no different - they are usually supplied by 5V and regulated down to 3.3V for GPS chipset and the mag sensor.  GPS is usually connected to a UART and the sensor to I2C ,  It's just two chips sharing the same housing and regulator but otherwise separate.
You should be able to find code for the UNO and I'm sure the part will run perfectly.  Bad chips are very rare and are usually unwittingly abused rather than "bad".   I also doubt there are many if any clones of these sensor chips.   

Cheers,
David

georgekucher, 2017-07-12:
Thank you David for your answers.

KiwiDavid, 2017-07-16:
Check out issue 1903 - just in case this relates to your board.

Cheers,
David

digitalentity, 2017-07-16:
@georgekucher cau you take a high-res photo of your compass chip? Would be good to check if it's indeed an HMC5883 or a pin-to-pin compatible part with incompatible software interface.

georgekucher, 2017-07-18:
[image]
[image]
[image]
[image]
[image]

giacomo892, 2017-07-18:
Also clean a bit the solder point :) That might help too! @georgekucher . Also those Flip32 AIO, which I had 2 of them, lasted like 10 minutes. I was even unable to make the first flight with them before the died.

georgekucher, 2017-07-18:
I have two of them as well. First I've damaged a bit (its UBEC is dead and I have use externall 5V) but it flights ok for several months. Second I use for flashing and connecting new devices.
As for my soldering skills, they are weak, but I have soldered & desoldered couple of times so won't help((

KiwiDavid, 2017-07-19:
Hi George,
Any chance of a direct close up looking down on the mag chip.  I think DE is trying to establish if your mag chip is a Chinese ring-in rather than the Honeywell part which is supported by the software.

Cheers,
David

georgekucher, 2017-07-19:
I'll make a photo in after 3 hours. I finally read 1903 post and I hope that there is the problem... So fingers crossed))

georgekucher, 2017-07-19:
[image]

georgekucher, 2017-07-19:
Hmm..
Seems like this is S883 or 5883 chip

georgekucher, 2017-07-19:
[image]
[image]
couple more

georgekucher, 2017-07-19:
May [this](https://github.com/mechasolution/Mecha_QMC5883) code help me?
It's from [here](https://www.reddit.com/r/AskElectronics/comments/5xo3md/problem_with_hmc5883l_magnetometer/)

Redshifft, 2017-07-20:
Yeah that's certainly not a HMC5883, I posted a picture here..https://github.com/iNavFlight/inav/issues/1903

What happened to the board ? It looks like you attacked it with a flame thrower :)

digitalentity, 2017-07-20:
@georgekucher this is not a HMC5883L, looks more like QMC5883:
[image]

digitalentity, 2017-07-20:
I'm closing this issue since we tracked it down to an incompatible sensor chip. Further discussion should happen in the issue about supporting that chip #1903. @georgekucher you are more than welcome to help with testing new code that we can start writing now :smile:

closed 2017-07-20 by digitalentity

georgekucher, 2017-07-20:
Thanks for all your help.
I'll be waiting go that feature to be ready to test. Besides I am ATQC, so testing is my job)))

Reyoking, 2020-05-29:
I bought a HMC5883l and it had the MAG=UNAVAILABLE problem.
got it working by changing the setting in configuration tab under magnetometer to QMC5883.