iNavFlight/inav #1864: confused when adding a sonar to an omnibus F3

ViciousPawn, 2017-07-05:
hi all!
 I have an hc-sr04 sonar module, that I want to add to my omnibus F3 running Inav. 
I'm running a gps module to uart2 and using an sbus receiver. 
I am also using uart1 for a bluetooth module.
Where should I connect trigger and echo of the sonar module?

DzikuVx, 2017-07-05:
Echo would be pin `RSSI` while Trigger would be `PPM input`. 
I do not think this hardware combination was tested, so you would be the first one

closed 2017-07-05 by DzikuVx

ViciousPawn, 2017-07-05:
Thanks for your reply.
The thing is... I have already connected my receivers signal cable to ppm input, as shown in the picture below..
[image]

DzikuVx, 2017-07-05:
Then, well... looks like the one who created the board did not thought this through and HC-SR04 is not usable on this board. Perhaps you can connect SBUS to UART2 RX directly and remove S.BUS jumper next to PPM pin. But I'm only guessing.

On the other topic, current sonar suport in INAV is rather problematic. It's used only during automated landings right now. And you can live without it. Also, like in your example, connecting HC-SR04 to most modern boards is not possible.

I'm slowly working on better rangefinder support for INAV, but that would take time.

ViciousPawn, 2017-07-05:
yes but then i would have to remove my gps module from uart2.
Well I guess it's not going to work on this board.
Thank you very much for your time, 
apreciate it.

DzikuVx, 2017-07-05:
You have 3 UARTs in total on that board. Should be enough to fit SBus and GPS and having PPM pad free. You might try, but like I said, current rangefinder support if highly limited