iNavFlight/inav #1866: Dual Receiver Mode? Both via MSP port and Transmitter / Receiver

johnsilvester, 2017-07-06:
I was wondering if it was possible to have two receivers connected to an SPF3 at the same.
I wanted to make an iPhone app controller controller via bluetooth and MSP but also have the ability to still use a regular transmitter like a FlySky. is this possible? Thanks!!

giacomo892, 2017-07-06:
Yes. When MSP data stop to be received it will jump to standard radio otherwise it will switch to failsafe

johnsilvester, 2017-07-06:
Thank you for the response! But I cannot get both PPM / PWM to work along side MSP due in the Configuration Tab under 'Receiver Mode', you have to select either 'PPM RX Input' , 'PWM RX Input', or 'MSP RX Input'.  Is there any way to select both?

closed 2017-07-16 by johnsilvester