iNavFlight/inav #1870: Omnibus F4 V2 S.BUS erratic behavior

Wimukthi, 2017-07-07:
Hi all,
[image]
[image]

I'm trying to run iNav on the Omnibus F4 V2 board which i bought from banggood (Link below) when I connect the S.BUS output from my Radiolink R9DS to the S.BUS pin and configure UART1 as serial and correctly select serial receiver on iNAV 1.7.1 i get random fluctuating values on the receiver tab, so far i couldn't figure out why this is not working, I used the firmware OmnibusF4PRO to flash to this board, maybe I used the wrong version? Any help would be much appreciated.
https://www.banggood.com/BF3_1_5-Omnibus-F4-V2-Flight-Controller-STM32-F405-MCU-Integrated-OSD-Built-in-5V-BEC-Current-Meter-p-1143258.html

DzikuVx, 2017-07-07:
do you have anything connected to PPM pin?

Wimukthi, 2017-07-07:
Hi DzikuVx R9DS doesn't have a PPM output, if you mean the PPM pin on the board, it's the same as the S.BUS pin right? At;east that's what I could gather from the information online. Btw the S.BUS output of the receiver works correctly, I tested it on a NAZA Lite

DzikuVx, 2017-07-07:
Do you have anything connected to UART1 headers? You connected SBUS to  SBUS/PPM pin, right?

Wimukthi, 2017-07-07:
No I have nothing connected to the UART1 header, I have the M8N GPS with Mag connected to UART6 and UART3

moggiex, 2017-07-08:
Daft one, did you disable MSP on UART1 and enable "Serial RX" on UART1 as well?

Wimukthi, 2017-07-10:
I've managed to correct this issue, it seems it was caused by the static electricity generated by the SMPS of my desktop PC.

closed 2017-07-10 by Wimukthi