iNavFlight/inav #1872: Not sure if bug: FAILSAFE "land" not descending when ALTHOLD is active.

darmach, 2017-07-07:
Few days ago I experienced what I'd consider a bug in failsafe behaviour. Not sure if it's relevant in light of https://github.com/iNavFlight/inav/issues/1475

I recently installed iNav, was testing ALTHOLD/POSHOLD. It was holding and then decided to test manually enabled (aux switch) FAILSAFE - configured to lower throttle and "land". Quad started beeping as expected, but did not descend. After configured 15 seconds for land elapsed, quad disarmed mid-air, falling.

Looks like enabled ALTHOLD stopped failsafe semi-land from descending. Shouldn't failsafe procedure be executed no matter what other mode is active? In this case I was low, and over a empty field, could've been worse.

I can replicate and attach logs if necessary.

digitalentity, 2017-07-08:
@darmach thanks for reporting. This bug indeed needs to be fixed ASAP.

Manawyrm, 2017-07-08:
@digitalentity: Can confirm, a friend had his copter falling out of the sky while testing failsafe in this exact configuration. 

./main/navigation/navigation.c:void activateForcedRTH(void) is responsible for this functionality. fallback RTH is the only place this method get's called. 

Something like
  
```
DISABLE_FLIGHT_MODE(NAV_ALTHOLD_MODE | NAV_POSHOLD_MODE | NAV_WP_MODE | NAV_LAUNCH_MODE);
ENABLE_FLIGHT_MODE(NAV_RTH_MODE);
```
at the top of this function should force the correct mode, right? I will test this and create a pull request.

digitalentity, 2017-07-08:
@Manawyrm @darmach please attach your configurations.

@Manawyrm it's more complicated than that. Navigation is handled via a state machine, not flight mode flags. Likely cause of the bug is handling the failsafe condition.

digitalentity, 2017-07-08:
Found the problem. Failsafe didn't override navigation system (PosHold/AltHold/RTH/WP) modes. This caused a bug when failsafe was applying control input for an open-loop landing, but AltHold was overriding it later in the code. 

Fix pending.

digitalentity, 2017-07-08:
Guys, please test https://github.com/iNavFlight/inav/pull/1873, failsafe LAND and DROP should now bypass AltHold and other navigation modes.

giacomo892, 2017-07-08:
Is this only referring to ones that set LAND/DROP action as failsafe?

I tested failsafe many times, having setted it as RTH, and always landed (and disarmed) the machine even if it was ALT+POS HOLD previously.

digitalentity, 2017-07-08:
@giacomo892 yes, only LAND/DROP procedure is affected by the bug. RTH requires navigation subsystem and therefore is not affected by the bug at all.

closed 2017-07-15 by digitalentity

darmach, 2017-11-20:
Should this issue still exist in 1.7.3? (I am aware that 1.8 is already out)

stronnag, 2017-11-20:
Am I losing my sanity .... 
````
grep -r --include \*.c bypassNavigation .
# nothing
````

digitalentity, 2017-11-21:
@stronnag at which version of INAV did we have "bypassNavigation" ?

digitalentity, 2017-11-21:
@stronnag I must be insane as well. How come #1873 was not merged?