iNavFlight/inav #1876: Feature Request: Rate/Angle hybrid mode

heathicus, 2017-07-10:
While training my young daughter to fly her drone FPV, I thought of a good feature that would be helpful.  I don't want to start her off with Angle mode and the bad habits that starts, but Acro is a little rough.  But if there was a mode that behaved just like Acro (no auto leveling) except limited the angle the craft could be in, like Angle mode, that would help significantly.

oleost, 2017-07-10:
Unsure what's going to happen with angle mode with angle p gain set to zero?

DzikuVx, 2017-07-10:
@oleost that would not work.
@heathicus I actually like the idea, since I also want to teach my daughter to fly :)

Redshifft, 2017-07-10:
@heathicus  you are assuming the younger faster uncluttered brain will struggle, just soften it up a bit and go find some soft grass :D

closed 2018-05-27 by stale[bot]