iNavFlight/inav #1877: I2C devices not working on Omnibus F4 pro

saswatahira, 2017-07-11:
Yesterday I got an omnibus f4 pro board from AliExpress. I tried to add external compass and GPS module (HMC5883L and Ublox Neo6M). fortunately the GPS is detected after connecting to the UART6, but the compass is not being detected after connecting to the UART3/I2C port and enabling serial RX on UART3. In fact, no external I2C device is detected on the I2C bus. I tried an external barometer and an external mpu6500 gyro (both different from internal sensors). There is 3.3v power in the I2C. Can anyone please help?

DzikuVx, 2017-07-11:
If this is Omnibus F4 Pro **v3** then this board is not fully supported. @jflyper you seems to have one of those v3 boards...

jflyper, 2017-07-11:
@saswatahira 
For I2C on J10 to work, you must disable everything on UART3; I2C and UART3 shares pins, and as soon as something is enabled for UART3, I2C will be disabled.

saswatahira, 2017-07-11:
Mine is omnibus f4 pro V2. I have not enabled UART3. I know they have shared pins. But how to use I2C only? Can I do it from CLI or do I have to compile the source code with disabled UART3 pins? I don't have any clear idea. Can someone please guide?

DzikuVx, 2017-07-11:
@saswatahira in that case I suspect hardware problems on your side. I can confirm Omnibus F4 Pro v2 working with I2C as long as UART3 is not enabled in Configurator

saswatahira, 2017-07-11:
[image]
[image]
[image]
First I tested a gy-86 sensor board with Arduino and multiwii, then connect the VCC, GND, SCL, SDA to the respective pins and enable the serial RX on UART 3 as seen in the photos. Am I doing anything wrong? Or is the board defective?

jflyper, 2017-07-12:
You have serial RX enabled on UART3. Disable it.

saswatahira, 2017-07-12:
sorry for the screenshot with the UART3 serial RX enabled. anyway, I was experimenting with multiple options. Problem was solved, seemed the Omnibus board and the sensor board both were OK, but the on-board mag sensor on the GPS module did not have some resistors on the output line. Testing with a single HMC5883L breakout board solved the problem. it had all the pull up resistors in the output line.

So in conclusion, iNav is OK, Omnibus F4 pro V2 is OK. the Compass with GPS board was partially faulty, which can be corrected by adding two resistors on the SCL and SDA pins to VCC. So, in my opinion this issue can be closed.

[image]

saswatahira, 2017-07-12:
[image]

Well I solved the issue by adding two 1K resistors between VCC to SCL and VCC to SDA. It works now.

DzikuVx, 2017-07-12:
So I assume we can close it, right?

closed 2017-07-12 by DzikuVx

FlyingGustel, 2017-08-12:
I can confirm this fix. It worked for my Omnibus F4 Pro + ublox m8n with HMC 5883.

yellowpanther, 2017-09-11:
@saswatahira you saved my day (and money ;))

Worked with this GPS unit (with HMC5883 mag) and my Omnibus F4 Pro (v3) :
[image]

Pokornz, 2017-11-25:
@saswatahira So even though the voltage on the line is 3.3V (you say that in the first post), you had to add 1k pullup resistors?

jflyper, 2017-11-25:
It is not only the voltage, but how fast the bus can recover the voltage from low state (driven by devices) to hi state (idle state). Larger (weaker) pull-up resistors make it slower, smaller (stronger) ones make it faster.

On the contrary, stronger pull-ups makes devices harder to drive the bus low, while weaker ones make it easier. Think of the case that the bus is tied to 3V3 supply (strongest pull-up); it could never be driven low.

I suggest that synthesized pull-up resistor value to be around R2.2K.

Ref:
http://dsscircuits.com/articles/effects-of-varying-i2c-pull-up-resistors

PScott1, 2017-11-29:
@saswatahira. Thanks I have the same older style M6 GPS/mag board (CJMCU-APM)  and was scratching my head why I couldn't get a i2c connection. 
@jflyper thanks for explaining about the pull up resistor. I had no idea.

asmeureanu, 2017-12-01:
@yellowpanther what resistors did you solder and where is best to do it? 
I am in the same situation as you were.

yellowpanther, 2017-12-02:
@asmeureanu I soldered 1k resistors between sda and vcc, and scl and vcc. I don't know where is the best to do it, but I did it direcly on the wires, with standard resistors (not smd) and it worked.

rholt51, 2017-12-15:
i'm surprised how LONG it took me to find this fix (which worked!, btw). i'm very happy and i'm giving myself some kudos for searching as long as i did to find this thread. i made perfect sense to me and i was encouraged by the people who tried this and succeeded. i, too, succeeded. thanks so much for the help on this. right now i have 1/4watt resistors 'cause that's all i had lying around, but i will be ordering some tiny 1k's tonight.
Russ from Coral Springs, Floriday, USA

tropnevad, 2018-02-27:
Have you tried to use resource mapping, as the omnibus boards (and many other FC) use configurable I2C ports so you need to map them using CLI

In the case of Omnibus F4 the following should help.. 

```
resource i2c_scl 2 b10
resource i2c_sda 2 b11
```

DzikuVx, 2018-02-28:
@tropnevad there is no resource mapping in INAV

simtcrom, 2018-08-21:
I am using a Readytosky Ublox NEO-M8N GPS Module w/ Built-in HMC5883L. My FC is Omnibus f4 pro v2. Initially the compass was not being detected correctly and I had reduced i2c speed to 200KHZ. That made compass detected. But during flight, it constantly stop functioning (blackbox logs).
And I think, it is due to this pull up resistor issue.
I have attached my current wiring diagram, can someone here please confirm if I have placed the resistors  in correct place?
[image]

digitalentity, 2018-08-21:
Are you sure that GPS module doesn't have it's own 3.3V regulator?

simtcrom, 2018-08-22:
I dont know that for sure. I am attaching a photo of my GPS module inside. Could you please check?
[image]

simtcrom, 2018-09-13:
I am all set adding pull up resisyors

teckel12, 2018-09-13:
Sounds like this is not an INAV issue, more like a Chinese clone issue due to lack of "unneeded" pull-up resistors.

yo8aiv, 2019-06-05:
thanks for posting sollutions guys.
mag issue solved by adding 2 pullup resistors (2.2k) to 3.3v inside gps

tijira-1999, 2020-04-21:
> I am all set adding pull up resistors

Could you please show your connection diagram

tijira-1999, 2020-04-21:
> I am all set adding pull up resistors

I have the exact same gps module please show your wiring diagram

Are you using 5v as VCC or 3.3v??