iNavFlight/inav #1878: Throttle telemetry feedback from actual motor output

wilco1967, 2017-07-11:
Feature request. (Frsky SPort, iNav 1.7.1)

Would it be possible to get the actual motor throttle, as sent out to the ESC, shown in the telemetry ? 
Right now, the telemetry is showing throttle stick position.

When flying in RTH or alt-hold (any mode where throttle is automatically controlled), it would make more sense to show the output to ESC, rather then THR stick position.

Thanks a lot !

digitalentity, 2017-07-13:
Telemetry sends rcCommand[THROTTLE] as a value. It's set by AltHold and it's exactly the value that goes to ESCs.

giacomo892, 2017-07-13:
@digitalentity I think on MWOSD for example you can only show the remote stick position only :) Maybe that's the issue.

wilco1967, 2017-07-13:
Yes, giacomo, that's correct. 

On MWOSD, but also on the build-in OSD (Omnibus F3), it always shows throttle stick position....
I might not have explained that clearly enough earlier. (I wrote telemetry first, before realizing I really meant OSD).

For external MWOSD, it probably a 'feature' in the MWOSD software.  
But for build-in OSD, it is part of iNav software (might have originated from the same code base perhaps ?).
Happens both on the build in and external OSD.

No big deal, but if it's a minor change, perhaps worth 'improving' (at least on the build-in OSD)....

Thanks !

fiam, 2017-09-17:
Since we already have this on the telemetry, it seems this would be completely fixed by fixing #2031.

fiam, 2017-09-26:
Closing this as a duplicate of #2031, since that one has some conversation regarding how to implement the feature.

closed 2017-09-26 by fiam