iNavFlight/inav #1879: 3D Neutral is not Symmetrical

KenImhof, 2017-07-11:
3D Neutral offset appears to be hard coded around 1550ms and cannot be changed. I have [Mid Throttle = 1500] and [3D_Neutral = 1500].  With [3d_deadband_throttle = 1], the motors change from reverse to forward around 1551ms. They then change back from forward to reverse around 1550ms as measured from the receiver window.  Changing the 3d_neutral value does not make any difference.

KenImhof, 2017-08-17:
Submitted pull request #2006.

KenImhof, 2017-09-11:
One very simple fix that is working for me and allows the center point to be adjusted is to change 
rxConfig()->midrc to flight3DConfig()->neutral3d on the void mixTable(void) procedure.

DzikuVx, 2020-02-26:
fixed in #5441

closed 2020-02-26 by DzikuVx