iNavFlight/inav #1882: No connection with SPRACEF3-MINI over USB

schuppenkarp, 2017-07-12:
I am not able to connect to the board with the newest Version 1.7.1 also older version starting with the 1.6 are not working. The USB VCP is correctly assigned to the flight control and the iNav Configurator also can open it but after 10 seconds the connection sequence ends because no configuration was received from the flightcontrol.

There should be no problem with the board because Cleanflight in the newest Version is working without any Problems on the same board ans same configuartion.

The User-LED is blinking on Startup 3xfast-pause-3x-VeryFast-Pause-2TimesFancy-pause-continouseBlinking

tonyintn, 2017-07-13:
The problem is not Inav many including myself has never had a connect problem with the SP F3 on any build.  
I would start by checking your USB dongle and making sure you setting are correct also might check to see if your USB connect is actually working i.e. missing drivers.

schuppenkarp, 2017-07-14:
Board and usb are working 

I can flash clearflight and everything is working 

I have the spracer f3 mini

tonyintn, 2017-07-14:
This is just a hunt in the dark but something to try; the mini has limited memory thus you could be running into a problem with that.  
Keep in mind iNav files uses more space than a cleanflight file as it has more going on internally.  
F3 built file has everything turned on and using the Normal F3 to the max; Rebuild your own F3 file but turn things off such as LEDs and/or blackbox this is just a test as you are probably maxing out the F3 mini.

schuppenkarp, 2017-07-17:
i think the mini version is using the same uC as the SPRacing F3 (STM32F303CC) so i dont think so that the problem is in the flashsize. Is there anywhere a documentation of the blinkcode at the startup of the FlightControl? Or is there a way to get out some Boot Messages on one of the serial ports to find the problem?

Best Regards

digitalentity, 2017-09-08:
Can't reproduce

closed 2017-09-08 by digitalentity