iNavFlight/inav #1883: Fixedwing going down in rth mode with allow_landing off

retif, 2017-07-12:
I have cc3d with 1.7.1 inav with gps. Allow_landing off, rth_altitude fixed 100m. When nav rth mode switched on - drone makes few circles at home, then with no other indication in osd it takes straight forward course and drops altitude, it looks like it just trying to land. Sometimes it even does not fly to home point and just trying to land.

My cc3d does not support blackbox feature

digitalentity, 2017-07-12:
Most likely you have marginal GPS quality and your plane enters emergency landing sequence when GPS is lost.

Faduf, 2017-07-12:
I tried the RTH mode last week with a fixed wing and Inav 1.7.1. It worked perfectly. Even without baro or magneto. Only GPS was used.

retif, 2017-07-13:
Can inav draw any indication on osd, that its going  to emergency landing?

retif, 2017-07-19:
tried other gps module, same problem

digitalentity, 2017-07-19:
@retif give us some details about your setup. Config dump, what GPS modules did you use, etc.

retif, 2017-07-19:
sorry, i am using 1.6.1 ppm1 cc3d firmware.
configuration: https://pastebin.com/A5gZqfQQ
gps: UBLOX 7 naze32 cc3d

digitalentity, 2017-07-19:
Please upgrade to 1.7.1. RTH landing bug was fixed in https://github.com/iNavFlight/inav/pull/1665

closed 2017-07-19 by digitalentity