iNavFlight/inav #1894: Led_strip on Hexacopter

michaelvdhoek, 2017-07-14:
Hi there, 

I'm building an Hexa X copter just for fun and to play around with the GPS modes. It's riggid with a Omnibus F4 V2 which has a dedicated LED_strip port. This port isn't used in the code because its redirected to PWM 5. 

But, on the Hex I also want to use the LED_strip! 
Can the code be changed so the led_strip feature is available for the omnibus users?

Cheers, Michael

DzikuVx, 2017-07-14:
Unfortunately, this LED pin is more less useless and can not really be used for LED strips. It's a hardware design fault

closed 2017-07-14 by DzikuVx

michaelvdhoek, 2017-07-14:
So HEX X and ledstrip isnt going to work at all?
Why is the LED_strip pin even there? I guess it isnt there for no reason?

DzikuVx, 2017-07-14:
I'm affraid not. 
This question you have to ask the manufacturer of that board