iNavFlight/inav #1896: HGLRC F3 V3.1 issues (SP Racing F3)

konraddi, 2017-07-14:
I bought a "HGLRC F3 V3.1" flight controller because I have made good experiences and many flights with iNav from version 1.5 to now on different boards (naze32, sp racing f3, Omnibus F3) and flying wings. 

So I flashed iNav 1.7.1 on the HGLRC board, and the newest mwosd 1710 on the integrated minimosd.

First problems I noticed were the incorrect OSD information. Speed, altitude infos and so on are delayed around 5-15 seconds, so I never had the actual speed or altitude values, but the info from several seconds before. But flying in acro or angle/horizon mode was possible.

When switching to any more complex mode (Althold, Poshold,  RTH) my wing is behaving extremely weird, and completely uncontrollable. Every second flight, when awaiting to a navigation mode,  the board shuts off and reboots completely, so I have to emergency land without motor because it is disarmed.

Is there a way to find out where the problems are  and how to solve them? I have the impression that the board is overloaded, but that should not be the case with F3 CPU and normal settings?! 

Thanks! 
Konrad 

Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# INAV/SPRACINGF3 1.7.1 Jun  3 2017 / 08:13:23 (48e1feff)

# resources

# mixer
mixer FLYING_WING

# servo
servo 3 1000 2000 1500 45 -1 
servo 4 1000 2000 1500 -45 -1 

# servo mix

# feature
feature -RX_PPM
feature -BLACKBOX
feature RX_SERIAL
feature MOTOR_STOP
feature GPS
feature CURRENT_METER
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 1 2 115200 9600 0 115200
serial 2 64 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1900 2100
aux 1 2 5 1400 1600
aux 2 20 0 900 2100
aux 3 3 0 1875 2100
aux 4 9 5 900 1125
aux 5 8 2 1925 2100
aux 6 10 0 900 1100
aux 7 27 4 900 1100
aux 8 28 4 1400 1600
aux 9 18 3 1900 2100

# adjrange

# rxrange

# master
set gyro_sync = ON
set gyro_hardware_lpf = 188HZ
set acc_hardware = MPU6050
set acczero_x = 217
set acczero_y = -40
set acczero_z = 209
set accgain_x = 4052
set accgain_y = 4072
set accgain_z = 3990
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SBUS
set motor_pwm_rate = 2000
set motor_pwm_protocol = MULTISHOT
set failsafe_procedure = RTH
set align_board_pitch = 65
set multiwii_current_meter_output = ON
set servo_pwm_rate = 330
set disarm_kill_switch = OFF
set auto_disarm_delay = 0
set gps_sbas_mode = AUTO
set deadband = 7
set yaw_deadband = 7
set nav_rth_climb_first = OFF
set nav_rth_climb_ignore_emerg = ON
set nav_rth_alt_mode = MAX
set nav_fw_cruise_thr = 1300
set nav_fw_min_thr = 1150
set nav_fw_bank_angle = 25
set nav_fw_climb_angle = 35
set nav_fw_dive_angle = 20
set nav_fw_loiter_radius = 8000
set attitude_task_frequency = 500

# profile
profile 1

set fw_i_pitch = 11
set fw_ff_pitch = 52
set fw_p_roll = 3
set fw_ff_roll = 35
set fw_p_yaw = 5
set fw_ff_yaw = 45
set fw_p_level = 8
set nav_fw_pos_z_p = 25
set nav_fw_pos_xy_p = 40
set nav_fw_pos_xy_d = 7
set rc_expo = 30
set roll_rate = 40
set pitch_rate = 30
set tpa_rate = 20

#

konraddi, 2017-07-17:
A quick update: The problems only exist in iNav 1.7.1!!

I flashed iNav 1.7.0 and the problems disappeared. Maybe it has something to do with the latest change "Fix board hangup when flooded with MSP requests"? I also tested the newer mwosd 1712, but no change, the error seems to the located only in iNav 1.7.1. What can I do to support in finding and fixing the error? iNav 1.7.0 lacks the new features for bank angle in ALTHOLD Mode...

Thanks,
Konrad

DzikuVx, 2017-07-18:
This is not officially supported board. The fact that is uses a SPRF target does not means it is SPRF3.

Without actually having the board by one of the devs it is rather impossible to determine the exact cause and fix it

konraddi, 2017-07-18:
Hi Paweł,

thanks for your quick response!  I am no developer so I only understand the most basic and simple things with this. 

The only explanation for me is a change in the firmware code from 1.7.0 to 1.7.1 because the board works with iNav 1.7.0. 

"Fix board hangup when flooded with MSP requests" sounds for me that a change was made which could possibly cause those issues in 1.7.1. 

Other than the added  OSD (separate MinimOSD) BEC PDB and Current Sensor the board should be really identical or is that only my wishfull thinking?

Could a Blackbox Log help for fixing it? My friends and I bought 4 of those boards in total for using it with iNav only. So when it is really needed for a dev to get hands on this board for fixing it, we would sacrifice one for sure.

DzikuVx, 2017-07-18:
@konraddi blackbox will not help. It looks like communication problem. I'm using MinimOSD over UART with one of my build and it is working just fine with all versions of INAV. 
We have no idea how this is wired so on. Or how it is polling data etc. Looks like you've hit an edge case over here.
You can try to compile your own 1.7.2 with that commit for MSP flooding reverted and test if that helps.

konraddi, 2017-07-18:
Dear Paweł,

it definitely sounds like an edge case. I am very interested in testing the newest iNav version but without that one MSP flooding change. Sadly I do not know how to compile. Could you do that for me or show me, so we can see if it works?

Many Thanks!
Konrad

DzikuVx, 2017-07-18:
OK, I've reverted 137fdc7f3a9dbb42eeeb0eb0bda910e3934678ef , here is a hex for SPRF3. You might give it a try

[inav_1.7.2_SPRACINGF3.zip](https://github.com/iNavFlight/inav/files/1156990/inav_1.7.2_SPRACINGF3.zip)

giacomo892, 2017-07-18:
I have a SPRF3 with an external MinimOSD and update rate is ok. Also I run telemetry/MSP on F1 with softserial and that commit saved my days :)

konraddi, 2017-07-19:
@giacomo892 you mean this reverted version "inav_1.7.2_SPRACINGF3.zip" works for you too? :)

giacomo892, 2017-07-19:
@konraddi No. I mean that 1.7.1 works flawlessy on my SPRF3 and NAZE too :)

konraddi, 2017-07-19:
@giacomo892 same here, the default 1.7.1 works good in my S800 and FT Spear on a normal SP Racing F3 from ebay with MinimOSD, but before the 1.7.0 worked as well.

Only the HGLRC F3 V3.1 seems to have that problem.

giacomo892, 2017-07-19:
Since the board is not supported maybe there is an issue with some timer and resource allocation there. @konraddi

digitalentity, 2017-07-19:
@konraddi this is likely hardware-related. That HGLRC F3 board probably is a knock-off of a popular SPRacingF3, but manufacturer must have added some incompatibilities to the hardware.

We can't troubleshoot bugs for board that is neither officially supported nor used by any of the deveoplers.

konraddi, 2017-07-19:
[image]

konraddi, 2017-07-19:
I wrote to the manufacturer 5 days ago. I will ask now if they could give some details about differences to the SP Racing F3 hardware

konraddi, 2017-07-19:
[image]

digitalentity, 2017-07-19:
@konraddi please let us know what they will reply to you.
If not, we can try doing a bisect for you, but it will take a while since we'll have to incrementally compile firmwares for you to test.

konraddi, 2017-07-19:
@digitalentity thanks for your reply, still no reply from the manufacturer.

But excellent news, the inav_1.7.2_SPRACINGF3.zip firmware works! I flew three packs today with it and had great fun.

One thing I noticed was that my wing flew very sluggish, and was way slower when rolling. I checked the servo throws in passthrough and acro mode, and they were much lower! 

So I had to up the servo rates from 45% to 75% to get the same throw as in iNav 1.7.0. What could be the cause for it?

Another quick question because it worked in iNav 1.7.0: RTH is shaking up and down, but in iNav 1.7.0 those PIDs were okay. In this video my wing flies with those PIDs. Then I lowered P value to 10 but only got minimal improvement. Should I up the I and / or D values? I do not understand completely what they do. With a P value of 5 and I and D the same, the wing is not correcting fast enough / still flies up and down most of the time and is unable to hold the horizon permanently.

https://youtu.be/UdQIIel2Jdg

[image]

With a big thank you!
Konrad

konraddi, 2017-07-20:
https://www.youtube.com/watch?v=qUxeQmCgZY8

konraddi, 2017-07-21:
How would you place the replies of HGLRC Company? Do they not understand what I write?

[image]
[image]
[image]
[image]

konraddi, 2017-07-21:
Yesterday they made a post in a Facebook Group to promote their direct purchase website / shop. I asked them about my problems because my mails to them were unanswered for over 6 days. Then today they deleted their post to make negative feedback and problems with their products disappear. What is going on here?

konraddi, 2017-07-24:
So I was unable to reproduce the problems with iNav 1.7.1 or 1.7.2.  It worked correctly without having to revert that MSP change.

Then yesterday I got the same issues on my Popwing 1300 with another SP Racing F3 Board that worked just fine in my FT Spear (http://www.ebay.de/itm/SP-Racing-F3-Cleanflight-Flight-Controller-Acro-6dof-for-FPV-Racing-Quadcopter-/322232324425?hash=item4b068b1149:g:FKkAAOSwHoFXtaAu)

I asked my buddy if he still had the same issues. He had, and hinted me to look at the GPS module. So i swapped the "new version" GPS to a older model that I still have. And there you go, evereything worked fine!

So the issue is with the "new version" GPS modules of this model and not the FC or iNav software: https://www.banggood.com/7M8M-Ublox-M8N-GPS-Module-for-APM-Pixhawk-CC3D-Naze32-F3-Flight-Control-p-1078657.html

[image]

Here are two videos, one with bad GPS module, one with the old working version of it:

https://www.youtube.com/watch?v=2u5nb1_yZO0
https://www.youtube.com/watch?v=dOqLE1qBpgw

digitalentity, 2017-07-24:
@konraddi omg, they seem to not understand what you are saying to them.
Regarding GPS module - I purchased this exact model to put on my wing a while ago and it worked for something like 5 minutes. Then it got extremely hot and stopped working. I guess there's some quality issue with that module :disappointed:

closed 2017-09-08 by digitalentity