iNavFlight/inav #1902: SP Racing F3 OSD not working with current OSD firmware

xtrmtrk, 2017-07-15:
If the SP Racing F3 OSD is upgraded to the latest OSD firmware (v2.0.5 - SPRACINGF3OSD - 09-07-2017 08:47 (stable)) it no longer works with iNav.  When you boot up iNav with 1.7.1 - SPRACINGF3EVO on the FC, it just displays "WAITING FOR FC."

Poking around in the OSD code a bit, it appears that it's not getting the MSP heartbeat.

xtrmtrk, 2017-07-15:
Please note that things more or less work with v1.14.2 of the OSD firmware.  Things seem to have stopped with v2.0.0

krzysztofmatula, 2017-07-16:
It seems that SPF3OSD evolve now together with Cleanflight and the compatibility with other controllers in not a goal. Unfortunately... 
A controller with a built-in OSD chip seems to be better option now.
A separate OSD board could be valuable thing if better, more powerful chip were used, e.g. capable of rendering smooth graphics. The SPF3OSD uses the same MAX chip that other OSD systems use, and it provides just a textual display with some possibility to play with fonts...

DzikuVx, 2017-08-09:
Unfortunately SPF3OSD is not part of INAV project as is not supported as a hardware. I'm affraid that we can not help over here. You should either use `v1.14.2 of the OSD firmware` or submit a bug to Cleanflight project which is managing this target

closed 2017-08-09 by DzikuVx