iNavFlight/inav #1903: QMC5883L compass support

digitalentity, 2017-07-15:
Today I received some info that there is a Chinese QMC5883L chip which is pin-to-pin compatible with HMC5883L and that it might be installed on some of boards that originate from China.

The thing is that QMC5883L has configurable at the factory I2C address so it might be configured to respond to same 0x1E address as Honywell sensor. However, the register map is **different**.

Default address for QMC sensor is 0x0D which might explain why on some boards/GPS units we can't get compass to work no matter what.

KiwiDavid, 2017-07-16:
Wow - this could be a can of worms unless people read the fine print.  It seems they are already prolific among the various little break-off boards and adapters.  I know at least one user who can't get his recognized by iNav and this could well be the cause.

I read one comment that said a Chinese company had licensed the manufacture of the obsolete Honywell part but used a different address and a few other register changes which I guess makes it void as a direct replacement.  None of this is easily verified or confirmed.

The worst that could happen is that these parts are considered fair game for a "running change" without notice in the manufacture of FC's and other modules.  Read the small print!

Cheers,
David

digitalentity, 2017-07-16:
@KiwiDavid there is also IST8310 chip which is pin-to-pin compatible with Honywell's polular compass but has a different address and different register map. Despite being pin-to-pin compatible neither IST8310 nor QMC5883 are a drop-in replacement for Honywell's part due to API differences.

I doubt Chines would list the compass chip model even in the fine print - they usually write "GPS + Compass module".

KiwiDavid, 2017-07-17:
I believe you're right but having a chip which is pin compatible means the manufacturers of the FC's and break-out boards don't even have to run a pcb revision.  They can  carry on without even blinking and may even get a modest price reduction with a different chip while passing on the real problem to software people like yourself.

Redshifft, 2017-07-17:
Very interesting and goes a long way to explaining Mag issues people have reported.

Now the Question is are there components out there labeled L8832105 that are not HMC5883L ? with the original Honeywell part being EOL for a long time who knows :) Also is that copy part marked QMC5883L ? it would make identification easier

I believe this picture is the original Honeywell Mag
[image]

digitalentity, 2017-07-20:
This is a picture of QMC5883 mag: 
[image]

digitalentity, 2017-08-29:
Code verified to work. Merge pending. Closing.

closed 2017-08-29 by digitalentity

MRX-PLOIT, 2022-09-23:
Every chip with "DA5883" label IS a qmc5883l the original hmc5883l chip has "L883" label