iNavFlight/inav #1904: Add I2C support for Matek F405

digitalentity, 2017-07-16:
Continuation of https://github.com/iNavFlight/inav/issues/1781

stronnag, 2017-07-17:
Board arrived (just in case some testing is needed). Very nice FC.

stronnag, 2017-07-20:
Received info from Mateksys. Working on the bench. PR to follow.
````
# status
System Uptime: 71 seconds
Voltage: 0 * 0.1V (1S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP085, MAG=HMC5883
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, GPS=NONE
SD card: Startup failed
Stack size: 2048, Stack address: 0x10010000
I2C Errors: 0, config size: 3314, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = none
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 1, cycle time: 1004, PID rate: 996, RX rate: 49, System rate: 9
````
Also tested with BMP280.

DzikuVx, 2017-07-28:
Closing since #1920 is in place

closed 2017-07-28 by DzikuVx