iNavFlight/inav #1910: SET_RAW_RC command ignored

gilklar, 2017-07-16:
It appears that MSP commands SET_RAW_RC (opcode 200) are ignored.

Enabling SERIAL_RX does not help. So does sending the commands repeatedly. 

I am able to control the motors using SET_MOTORS command (opcode 214 ), and read ATTITUDE for 

example, so my COMM is OK.  However, SET_RAW_RC does not seem to work.

I am running latest 1.7.1

johnsilvester, 2017-07-16:
How are you setting these values? and where are you connecting it to in order to send the MSP commands?
IF USING UART:
Do not enable SERIAL_RX but enable MSP on the respective UART port. Note: keep MSP enabled on UART1 as well.

gilklar, 2017-07-16:
I tried both UART1 and UART2, with and without SERIAL_RX enabled. In all cases the commands were insuccessful. Again,  other commands work well. Could it be related to flight mode? I am trying it under HORIZON.

johnsilvester, 2017-07-16:
Mind doing a CLI Dump?

stronnag, 2017-07-16:
Works fine here. If you're using a F1 target (which is my bench test), you need to compile your own firmware with
````
#define USE_RX_MSP
````
There is enough space, you don't need to disable anything.
Then, with (inter alia):
````
# diff
# INAV/NAZE 1.7.2 Jul 16 2017 / 18:03:44 (ab01337e)
feature -RX_PPM
feature RX_MSP

serial 0 1 115200 38400 115200 115200
serial 1 1 115200 38400 0 115200
````
Running the configurator on /dev/ttyUSB0 (USART0) and the script at  https://transfer.sh/IEUI0/msp_set_rc.rb on /dev/ttyUSB1 (USART1), the receiver bars in the configurator dance around as expected. 

Probably in pretty much the same way as the last time someone claimed it wasn't working.

digitalentity, 2017-07-17:
F1 targets (NAZE, CC3D) don't have MSP RX support by default. You need to compile your own firmware to be able to enable `feature RX_MSP`

closed 2017-07-17 by digitalentity