iNavFlight/inav #1913: Can't connect to gui without battery connected after switching to external mag

FranzAr, 2017-07-17:
I removed the the mag from my sp racing F3 and connected the external mag from my M8N GPS.
Now when no battery is connected (and mag is not powered up) I can't connect to the FC from the iNav GUI. It seems like CPU load is going to high,

When battery is connected everything works as expected, mag works fine inflight, no issues.

How can I solve that?

giacomo892, 2017-07-17:
You need to connect the mag to the I2C port picking 5v from it so it gets powered when you plug the USB port otherwise you get stuck in the boot process.

KiwiDavid, 2017-07-17:
I don't think that would be possible because the GPS chip and the mag likely share the same 3.3V regulator.
The GPS/mag device is powered off a 5V supply but internally regulated down to 3.3V.  The mag sensor is not rated for direct 5V operation and the I2C bus is expecting 3.3V referenced pull-ups to interface to the micro.

FranzAr, 2017-07-17:
@giacomo892 
Thanks, I thing, that will solve my problem as the I2C 5V out is powered even by USB. Great, didn't know that. So I'll power my GPS/Mag thru the I2C 5V.

Thank you so much :-)

giacomo892, 2017-07-17:
Let us know and then close the issue :)

FranzAr, 2017-07-17:
Works perfectly, thanks for your help, guys :-)

closed 2017-07-17 by FranzAr