iNavFlight/inav #1919: Change sensor adress in configurator ?

GTorreil, 2017-07-20:
Hello, 

I need iNav to use the MPU9250 at its 'high' address. 

-> Maybe address selection for the sensors could be implemented in the configurator ?

Setup :

(SPF3) <--- I2C ---> (MPU9250 Module)

There are 4,7K pull up resistors on both i2C lines.

I had to change the MPU9250 address to get the FC to boot, probably because the (onboard) MPU6050 uses that same address. I used the address selection pin as it is stated in the datasheet :

[image]

In the meantime I need iNav to find the MPU at that address (b1101001) and not the default one (b1101000). How should I proceed ? — If I want to use only the magnetometer of the MPU9250 is there a better way of doing things ?

DzikuVx, 2017-07-20:
This will not happen before F1 targets are dropped. Then, maybe...

digitalentity, 2017-07-21:
@GTorreil MPU9150 is a combination of MPU6500 and AK8963 on the same I2C bus, but you can't access the compass chip unless you initialize the gyro and enable passthrough. In current state of development you either have to use the whole MPU9150 or not use it at all.

Implementing a "hacky" solution that will initialize the gyro on another i2c address and then initialize the compass is a very unique case, which I don't think we should be supporting officially. I might be wrong though.

GTorreil, 2017-07-22:
Ok, thank you 
Dropping F1 is something I am absolutely for :) but .. anyway. 
So if I understand well, removing the onboard MPU seems to be the only “clean” solution to use that external mpu-mag combo currently. Am I right ?

GTorreil, 2017-07-25:
SOLVED. This issue can be closed :)

closed 2017-07-25 by digitalentity