iNavFlight/inav #1928: SP RACING F3 HEX6X mixer not working in iNAV 1.7.2

jstremmler, 2017-07-23:
Just upgraded a hexacopter from iNav 1.6.1 to 1.7.2. FC is a SPRACINGF3.

When I specify the mixer as a Hex X in the configuration menue it returns after rebooting to a QuadX mixer.

 Same happens, when I key in in the CLI menue: MIXER HEX6X and SAVE. It returns to a QUADX mixer.

Is there any solution or do I have to operate with a custom mixer?

KiwiDavid, 2017-07-23:
Wouldn't that be a Hex Y?

jstremmler, 2017-07-23:
No, pls type in the CLI menu : "mixer list" to see all valid mixer commands! 
For a Hex X multicopter the correct command is: mixer  HEX6X and then save, but it returns always to a QuadX mixer.

jstremmler, 2017-07-23:
Will try a custom mixer today with:

mixer custom
mmix reset
mmix 0 1 -0.49 0.872 1
mmix 1 1 -0.51 -0.86 1
mmix 2 1 0.488 0.824 -1
mmix 3 1 0.47 -0.835 -1
mmix 4 1 -0.958 0.011 -1
mmix 5 1 1 -0.011 1

digitalentity, 2017-07-23:
Fix ready, closing

closed 2017-07-23 by digitalentity

dimak771, 2017-12-07:
Hello!
I have iNav 1.7.3 on CC3D Atom. I can`t use custom HEX6X in CLI:
# mmix 4 1.0 -1.0 -0.0 -1.0
index must be between 0 and 3
Can I configure HEX6X on 1.7?

stronnag, 2017-12-07:
https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20CC3D.md