iNavFlight/inav #1932: Smartport telemetry on SS in SPR Evo

BbypK, 2017-07-24:
I can't get smartport telemetry work over softserial in spracing f3 evo.

RX and TX are shorted and connected to motor outputs 7 and 8. Tried flashing both targets, SPRacingEvo, using Softserial2 and SPRacingEvo1SS, using Softserial1, none works.

If I connect to a real uart TX pin, smartport works.

Using latest version of Cleanflight and using same softserial config ( motor outputs 7 and 8, shorted, and selecting Softserial2 for Smartport telemetry), works without issues.

BbypK, 2017-07-27:
Using a 1N4007 diode between SS-TX and SS-RX did the trick. 
Diode was not necessary in my SP Racing F3 boards or the Evo board with cleanflight.

digitalentity, 2017-07-27:
Interesting. I'll look into how TX/RX pins are initalized.

BbypK, 2017-07-31:
After testing with two boards, and two receivers with the diode on smartport ( XSR, both updated to 170619 firmware ). I get sensors discovered and updated, but only for a short amount of time, varying from 20 seconds to 1 minute. Then, its like softserial port was deactivated, Taranis can't detect any sensor or update the values for sensors already discovered.

`Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# INAV/SPRACINGF3EVO_1SS 1.7.2 Jul 20 2017 / 11:47:45 (81b85de)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature RX_SERIAL
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature LED_STRIP
feature DASHBOARD
feature BLACKBOX
feature TRANSPONDER
feature PWM_OUTPUT_ENABLE

# beeper
beeper RUNTIME_CALIBRATION
[... 428 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 100
set rc_yaw_expo = 70
set thr_mid = 45
set thr_expo = 50
set roll_rate = 80
set pitch_rate = 80
set yaw_rate = 65
set tpa_rate = 10
set tpa_breakpoint = 1600`

I will test with one X4SB receiver that i have over here.

Zrelos, 2017-08-08:
I have same problem with SPRacing F3 X8R smartport and telemetry.
I tried with  1N4007 diode but without success.

With older version of INAV and one other SPRacing F3 everything had worked okay.

digitalentity, 2017-08-08:
@Zrelos is it also related to SoftSerial?

Zrelos, 2017-08-08:
Yes I tried with both SoftSerial 1 and SoftSerial 2

digitalentity, 2017-08-08:
Ok. Need to set things up and test.

BbypK, 2017-08-08:
Just tested on SPR Evo and X8R.
On both targets, Evo and Evo1SS, using Inav 1.7.2 i have telemetry working over Softserial using the diode.
Nothing attached to board except receiver.

BbypK, 2017-08-08:
And also tested on SPR F3, Inav 1.7.2 and X8R. 
Telemetry works on Softserial 2, with TX and RX shorted without diode.

Zrelos, 2017-08-08:
This is settings, what diode you use?

ÿ¾ÿ
Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# INAV/SPRACINGF3 1.7.2 Jul 20 2017 / 11:45:09 (81b85de)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature RX_SERIAL
feature MOTOR_STOP
feature SOFTSERIAL
feature TELEMETRY
feature PWM_OUTPUT_ENABLE

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
[... 423 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 50
set pitch_rate = 50
set yaw_rate = 45
set tpa_rate = 0
set tpa_breakpoint = 1500

#

BbypK, 2017-08-08:
You are using a SP Racing F3 board instead of Sp Racing F3 Evo, so diode is not necesary at least in my test with that board.
Just IO_2 pins 5 and 6 shorted together to Smartport pin on X8R and enabled Smartport Telemetry on SoftSerial 2.

Zrelos, 2017-08-08:
@BbypK  sorry, bad cable connection. Now everything work fine.

DzikuVx, 2017-08-09:
If it works, then I'm closing then. Thanks

closed 2017-08-09 by DzikuVx

teckel12, 2017-08-11:
I'm not sure this is resolved.  I have the same situation and I've redone the connection a couple times.  I'm also using SP Racing F3 (non-EVO) with the X4R-SB Rx and using soft serial connection with pins 5 & 6 tied together.

It won't work at all, and then work a few times in a row.  If I can get it running and get a GPS fix quickly enough and launch, I'll get the telemetry for the entire flight without a problem.  But, if it takes a while to get a GPS lock, in almost every case telemetry will drop.  I've resorted to the following procedure, connect a bench battery, power up, wait for GPS fix (the telemetry will drop after a few seconds).  I then disconnect the bench battery and connect a flight battery.  Because it previously got a GPS fix, it will get a GPS lock fast enough (typically) for me to launch and get telemetry.  But, this is still hit or miss.

Something just isn't right.  Not sure what the problem is.  I may try to connect pins 5&6 to a serial port and read them on the PC to maybe find out what happens.  But I'm at a total loss as everything is setup correctly (and working for entire flights when it does work).

BbypK, 2017-08-12:
Have you tried to backup your config and flash with other firmware like cleanflight and test if telemetry do the same ? Maybe it's not Inav related problem with that board.
I just tried two clones of SP Racing F3 (non Evo ) boards from Banggood Europe and get telemetry working over softserial is just imposible. Tried with X4R-SB and X8R, no one send data to radio.
Using same receivers and cables with and old retired SP Racing F3 board, works perfect.

Zrelos, 2017-08-13:
@teckel12 
The same problem is with my telemetry, sometime working, sometime not working.

teckel12, 2017-08-14:
It works, just not reliably. It appears to be a timeout issue, and if you can get into the air before one or more of the telemetry ports time out, it will work perfectly for the entire flight.  It's only a problem if it's not armed quickly enough (at least for me).

I have created my own build and will try a few things to track this down.

teckel12, 2017-08-14:
@Zrelos Are you using softserial?

Zrelos, 2017-08-15:
@teckel12

Yes softserial 2, sometimes works when I plug on off few times battery.
I thought it was a bad cable connection but It is not.

digitalentity, 2017-08-16:
There is a known issue with SoftSerial having issues with high baud rates. It's more evident on slow targets and less evident on faster targets but it's always there. S.Port uses 57600 baud which may be causing unstable behavior of softserial. @teckel12 would be interesting to know if you manage to find the issue.

teckel12, 2017-08-16:
@digitalentity @Zrelos As this issue is closed and I've created a duplicate issue I'm going to just point to the new issue instead of having two conversations about the exact same topic.

New issue to continue this discussion at #1990 (Smartport telemetry on SS in SPRacing F3)