iNavFlight/inav #1933: Omnibus F3 OSD issues in iNAV 1.7.2

Nihilistic, 2017-07-25:
The crosshair blinks on and off for me on the OSD with iNAV 1.7.2 loaded on my Omnibus F3 V3 with BEC.

Also, there is no "mph" label for the GPS speed when Imperial units is selected.

DzikuVx, 2017-07-27:
Freshly tested on Omnibus F4 v2 Pro and I confirm issue is there. Crosshair is blinking when AHI is enabled.
@marbalon I suspect this is connected with OSD refreshes. Looks like crosshair has to be rendered always after AHI

digitalentity, 2017-08-06:
Crosshair issue fixed. As for mph label - we lack the symbol in the font.

Nihilistic, 2017-08-07:
Sweet! So the crosshair fix appears in 1.7.3? Or 1.8?

digitalentity, 2017-08-07:
@Nihilistic crosshair fix will be included in next release. I'm not sure if it's going to be 1.7.3 or 1.8 yet. If we have new MSP stuff merged - probably 1.8

fiam, 2017-08-24:
@digitalentity Can you assign me this one? I'd like to get some OSD improvements done after #2019 is merged.

fiam, 2017-09-07:
This should have been automatically fixed when #2048 was merged. Can someone mark it as fixed manually?

PR #2048 (2017-09-04): Update SYM_MPH to use the new mp/h char rather than a blank
Requires the new font that will be available in the next version
of the configurator (already on master).

Fixes #1933
References #1756
commit cc465b10 (2017-09-04): Update SYM_MPH to use the new mp/h char rather than a blank

Requires the new font that will be available in the next version
of the configurator (already on master).

Fixes #1933
References #1756
closed 2017-09-07 by martinbudden