iNavFlight/inav #1937: iNav 1.7.2 ANYF7 with external Baro.

flyfisher2, 2017-07-26:
Using ANYFCF7_EXTERNAL_BARO, an external MS5611 baro is not detected.
Monitoring the external bus, as far as I can see, firmware never polls the external bus for a baro (but I could be wrong).
Is there something I am missing to make the standard release automatically detect an external baro?

FYI:
Looking at target.h in  ANYFCF7_EXTERNAL_BARO, I can see the ifdef that should enable the external bus for a baro.
Replacing that with the following works perfectly (although only for this specific hardware set up).

#define BARO
#define USE_BARO_MS5611
#define BARO_I2C_INSTANCE I2C_DEVICE_EXT

Gavsv8, 2017-07-27:
Have you got the other pins wired to select the correct address, CSB=low  and I2C mode PS high and have pull-up resistors in circuit?

[image]

DzikuVx, 2017-07-27:
@flyfisher2 I'm not following your logic here. Are you sure that you used and/or compiled `ANYFCF7_EXTERNAL_BARO` target, not generic `ANYFCF7` target?

flyfisher2, 2017-07-28:
Thanks for the responses.
Sorry I did not make myself clear .The standard hex worked fine but I self compiled 1.7.2 to move the buzzer to output 8 on the bg pro version and that was the problem.
It looks like I must have used the wrong target as everything is now good.
Very embarrassed.

closed 2017-07-28 by flyfisher2

DzikuVx, 2017-07-28:
No problem, good that it worked in the end