iNavFlight/inav #1938: Log raw sticks to blackbox

digitalentity, 2017-07-27:
Currently we are logging `rcCommand` which may be overriden by navigation. To troubleshoot some issues it's more important to have raw user input logged.

Possible solutions:

1. Add `rcData` to blackbox slow frame. Use reduced resolution and reduced rate of update for stick values so S-frames are not written too frequently.

2. Log raw sticks instead of `rcCommand`. Log `rcCommand` at reduced resolution in separate variables.

martinbudden, 2017-07-27:
Other possibilities:

1. Add conditional `rcData` blackbox field. Make `rcCommand` field conditional. That way user could choose which to log (or both).
2. Add debug mode so that `rcData` is stored in debug field and logged that way.

digitalentity, 2017-07-27:
Some things to consider: 

* `rcData` is updated relatively slow (compared to loop rate) - usually 50Hz compared to 1kHz (typical)
* `rcData` doesn't change often
* `rcCommand` may change each loop iteration when driven by navigation system, however usually limited to 50Hz
* In most cases BB log is required to do post-mortem analysis of a failure, therefore we should have logging of both `rcCommand` and `rcData` by default - it might not be possible to reproduce the failure with enabled logging of both.

martinbudden, 2017-07-27:
Note that slow frame interval is 4096, so it is probably too slow for `rcData`.

digitalentity, 2017-07-27:
@martinbudden it's either "every 4096 iterations" or "if data changed" - whichever comes true first

martinbudden, 2017-07-27:
Yes, I forgot that.

Note also that if a `rcData` does not change, in a blackbox P-frame the four zero-delta values can be stored in 1-byte. So cost of permanently saving `rcData` is of the order of one byte per P-Frame and perhaps 4 to 8 bytes per I-Frame.

Note also recent changes in Betaflight have reduced the frequency of I-Frames.

digitalentity, 2017-07-27:
@martinbudden true. This needs to be explored carefully. It might be enough to reduce rcData resolution to 1 byte so all of them could be stored in 4 bytes at most.

digitalentity, 2017-10-11:
Fixed by #2311

closed 2017-10-11 by digitalentity