iNavFlight/inav #1939: Omnibus F7 LED strips does not work

DzikuVx, 2017-07-27:
LED strips are not working. Looks like DMA issue. In theory it should be DMA2 Stream 4 but as soon as I enable LEDs on this DMA controller VCP stop to work while board seems to be booting.

flyfisher2, 2017-07-28:
iNAV1.7.2 LED strip also does not work on ANYFCF7.
OK in 1.7.1.

DzikuVx, 2017-07-28:
@flyfisher2 if so, please create a separate ticket. Those are probably different issues

flyfisher2, 2017-07-28:
OK will do.

digitalentity, 2017-09-08:
@DzikuVx is this still relevant?

DzikuVx, 2017-09-09:
When I checked 2 weeks ago, LED strip on Omnibus F7 was still not working. I did not checked after F7 LED fixes from @sambas . This board is currently assembled so I might be able to test next week

VoicOfReason, 2017-11-02:
LED strips are not working on my Omnibus F7V2 board.

shellixyz, 2017-11-16:
I had the same problem when making modifications on the PIKOBLX target (see #2101). I never used the LED_STRIP feature with iNav so I didn't know if it was working or not on other targets and assumed it could be related to the PIKOBLX target specifically. The symptoms were the same: VCP stops working as soon as LED_STRIP is activated and the board rebooted. Does somebody know if the LED_STRIP feature is working in v1.7.3 on F3 targets ?

steviegeek, 2018-05-01:
Omnibus F7 V2
Configurator 1.9.3
Firmware: 1.9.1 - OMNIBUSF7V2
Believed genuine, purchased from Unmanned Tech this week
LED_STRIP still shows 0 LED's in configurator tab
No LED listings in CLI Dump
Any news when this will be resolved ?
Thank you, Steve