iNavFlight/inav #1940: Allow navigation modes to override MOTOR_STOP

digitalentity, 2017-07-27:
If manual RTH is enabled with throttle stick at zero and MOTOR_STOP enabled - quad will fall out of the sky. For plane it's also a certain crash if pilot won't open throttle.

For safety reasons we should allow at least RTH and WP to override MOTOR_STOP

oleost, 2017-07-27:
Motor_stop and Althold also needs to be discussed.

My the problem I think is:

For airplanes it's natural to use motor_stop and is really the recommended way to have it.

For Multirotor motor_stop is just a stupid idea. 
Only reason for having it is if you lack and arm switch, but still then you have stick disarm.
Example DJI drones propellers always spin when arm, and they only have stick disarm combo.

marbalon, 2017-07-27:
Now if I enable RTL on plane with MOTOR_STOP and throttle=min plane goes crazy, don't turn correctly etc. If this is possible will be really useful make navigation works but if user keep throttle below min_check just turn off ALT hold but try to fly to next WP. It will be a great feature to gliders.

fiam, 2017-07-31:
From my reading of the code, it looks like this could affect failsafe when it's set to RTH, so this should be fixed (or mitigated) ASAP.

A plane with MOTOR_STOP gliding with THROTTLE <= min which goes out or the RC receiver range will crash, since failsafeProcedureLogic_t for FAILSAFE_PROCEDURE_RTH sets the THROTTLE channel to FAILSAFE_CHANNEL_HOLD. See https://github.com/iNavFlight/inav/blob/master/src/main/flight/failsafe.c#L117

Dronek, 2017-08-01:
> For airplanes it's natural to use motor_stop and is really the recommended way to have it.

I agree but why do we have to additionally set a parameter like motor_stop to have natural behavior of throttle control on fixed wings? I know, iNav was intended for multirotors first, but...
The way I see it, spinning motors on zero throttle is a feature for multirotors and motor_stop is a parameter to **override** this preset feature. It would make more sense if stopped motors on zero throttle were default behavior and instead of having motor_stop parameter we would then have e.g. "idle throttle" parameter to set for multirotors. This way there would be nothing to **override** throttle values in nav modes by moving sticks (other than disarming).

Xe54, 2017-10-04:
> For Multirotor motor_stop is just a stupid idea.
I can not agree with you @oleost . It is a very impressive maneuver to stop the motors during the flight and let the multicopter fall for few seconds. But it have to work reliable.

DzikuVx, 2017-10-05:
I have a crazy idea: let's get rid of MOTOR_STOP feature and assume it is always disable for MR and enabled for FW? Less is better and will solve a problem of users enabling it when they should not. Not only in NAV modes, but as a general rule

fiam, 2017-10-05:
@DzikuVx I'd totally support the idea of getting rid of the setting on FW and assuming it's always on.

However, there are use cases for enabling MOTOR_STOP on MC. For example, in small models for indoor use. Not that I think we shouldn't remove it on MC (you're probably better off with ßF than INAV on an small indoor model), It's just that I'm not as sure as in the FW case.

fiam, 2017-10-05:
I also wrote this in the PR, but it seems the bulk of the discussion is going on on this issue, so I'll replicate it here.

Mostly agree with ignoring MOTOR_STOP in RTH and WP modes. I know that it complicates the documentation a bit more, but I think we should always respect MOTOR_STOP when using stick arming or auto arming. Otherwise there's no fast way to turn off the motor[s] when shit hits the fan (no pun intended).

digitalentity, 2017-10-06:
Automatic modes (RTH, WP) do not accept user input. Why should throttle stick being at zero matter more than it being at 50% or 100%?

fiam, 2017-10-06:
@digitalentity It's not about user input, it is about having a fast "oh shit" mechanism when something goes wrong. Without an arm switch nor kill switch, there's no way to quickly stop an aircraft flying itself which is headed for some trees or a building other than enabling MOTOR_STOP and lowering THR. I think we should keep that functionality.

We could just add a function `bool hasQuickStopSwitch()` in `rc_modes.h` which would return true when using switch arming or having a kill switch mode configured. Then just change the PR:

```c
bool userMotorStop = !navigationIsFlyingAutonomousMode() && ...
```

To

```c
bool userMotorStop = (!navigationIsFlyingAutonomousMode() || !hasQuickStopSwitch()) && 
```

fiam, 2017-10-06:
Just for completeness, if it was up to me I'd remove stick arming and auto arming completely. They feel very unsafe to me. However, if we're keeping them we need to keep at least some safety measures for those use cases too.

stronnag, 2017-10-06:
Please do not even consider removing stick arm / disarm; my muscle memory will never forgive you. 

And I have this recurring nightmare about accidentally knocking the arm switch.

digitalentity, 2017-10-07:
All this is making configuring very tricky and somewhat inconsistent. Personally, I'm ok with how it works now - motor stop always respects user input (given there is valid user input = ! failsafe). However people are reporting issues with RTH falling out of the sky if throttle stick position is at zero (which by the way is a valid behavior IMO).

Let's discuss futher if we want nav modes to be able to override motor stop at all.

Dronek, 2017-10-07:
> Let's discuss futher if we want nav modes to be able to override motor stop at all.

How about working with existing parameters, one of which is **min_throttle**.

Please correct me if I'm wrong but if we set min_throttle to 1000 motor(s) will stop when throttle stick = min in all modes where throttle is actively controlled by the pilot even without MOTOR_STOP enabled. However, nav_modes would still rely on **nav_fw_min_thr**.

So with min_throttle = 1000 and MOTOR_STOP = OFF it should still be possible to stop the motors (for safety or for gliding) in non-nav_modes but it would prevent you from stopping the motor in nav_modes unless aircraft is disarmed.

I'm still against including an additional override for MOTOR_STOP since MOTOR_STOP already acts as an override (see my first post).

Dronek, 2017-10-08:
> So with min_throttle = 1000 and MOTOR_STOP = OFF it should still be possible to stop the motors (for safety or for gliding) in non-nav_modes but it would prevent you from stopping the motor in nav_modes unless aircraft is disarmed.

Works perfectly on the bench (with min_throttle = 1005, however). 
Motor stops at zero throttle in ACRO, ANGLE, PASSTHROUGH.
Motor keeps running at zero throttle in ALTHOLD.
I'm going to test this with RTH etc. in the air today.

digitalentity, 2017-10-10:
Another option is to remove MOTOR_STOP as a feature and make it a BOX mode assignable to a switch. This way user will be able to chose his own flexible behavior.

oleost, 2017-10-10:
Great idea, not necessarily new user friendly but gives a lot of flexibility.

digitalentity, 2017-10-10:
It won't change current behavor if you use it as a feature, but using it as a mode will allow user to disable it for althold/RTH. I agree it's not 100% user friendly, but it will give users an option to configure it to their liking.

Dronek, 2017-10-10:
> It won't change current behavor if you use it as a feature, but using it as a mode will allow user to disable it for althold/RTH. 

Sounds good and most flexible.

Do you think I'm running into any issues with my approach above? If not, I'll just stick with it because it seems to do exactly what I need.

digitalentity, 2017-10-10:
@Dronek if you keep `nav_fw_min_throttle` high enough - I don't expect issues.

digitalentity, 2017-10-10:
After some discussion on Slack - closed via #2259

PR #2259 (2017-10-10): Disable user-activated motor-stop if navigation system is doing RTH or WP
If navigation is flying autonomously - disregard zero throttle stick position in scope of MOTOR_STOP. This will ensure that user-activated RTH will always have control over motors even if throttle stick is at zero.

Fixes #1940
closed 2017-10-10 by digitalentity