iNavFlight/inav #1941: Position hold issues, quad flies in last direction input

takeshisama, 2017-07-27:
Hopefully this is the right place for this, please let me know if I should post this elsewhere.

I've been working on my first GPS-equipped quadcopter after many micros and racers, using an Omnibus F3 and a neo-m8n GPS.  I have done all calibrations several times over and, using EZ-GUI, confirmed GPS lock and location on the map, heading when pointed north (align_mag = CW270FLIP), and all that fun stuff, however I have never been able to get position hold to work.  When I first turn it on, there's a gradual drift that just gets faster and faster, if I try to correct at all then right after I let go of the right stick (mode 2), the quad levels out for a moment, then goes in whichever direction I last input, but much faster, and again doesn't seem to ever stop until I manually input the opposite direction. 

 I have a short flight's [blackbox log](https://github.com/iNavFlight/inav/files/1181330/LOG00004.TXT) and there's a video of that flight from 3rd person view [here](https://youtu.be/XZALYrnTjO0).  Let me know if there's anything else I can get, it's raining intermittently here but I should be able to do any tests necessary between it all.

Ralf-W, 2017-07-29:
Your blackbox log shows the number of sats jump between 13 an 0. There might be a hardware problem like a cold solder joint.
You wright "... and all that fun stuff, " and that your heading is reading 0/360 when pointing to the north. Which heading is displayed in the SETUP tab of the configurator when your quad is pointing East? It should read 90°.

stronnag, 2017-07-29:
0 sats oscillation is almost always a hardware (intermittent power) problem, like a bad joint. If you're using the compass on the GPS, you cannot depend on it reading correctly after this has happened.

takeshisama, 2017-07-29:
I'll work on redoing all soldering at some point tomorrow then, didn't think there was an issue since it let me take off and the map location was correct.  For heading when facing other directions, that seems to be completely accurate when viewing the heading numerically.  If I turn on the map display in EZ-GUI, rotating the craft rotates it on the screen and moving with it moves it as well.  I'll report back after cleaning up everything.

DzikuVx, 2017-08-09:
@takeshisama I'm closing this for now. If you find any new data that indicates this is software problem, please reopen. Thanks

closed 2017-08-09 by DzikuVx