iNavFlight/inav #1942: INAV - AnyFC7 upload not working

ThomasH3, 2017-07-28:
Hi

When I try to Flash INAV on my ANFC f7 Pro the tool is freezing during erase.
Betaflight and cleanflight are working.

flyfisher2, 2017-07-28:
untick FULL CHIP ERASE.
Should now work.

ThomasH3, 2017-07-28:
unfortunately, it doesn't work

DzikuVx, 2017-07-28:
@ThomasH3 which version of Configurator do you have?

ThomasH3, 2017-07-29:
Version 1.7.2
I was able to flash it. but Firmware will not start after reboot .... the connection closes after 10 sec

DzikuVx, 2017-07-29:
I just flashed AnyFC F7 Pro from BG with INAV 1.7.2 and no problems detected. Are you sure you are using latest C
[image]
onfigurator?
[image]

Could you try flashing the board using different PC?

ThomasH3, 2017-07-29:
After I disconnected my external GPS and compass, INAV is working now, however GPS, I2C (compass) and voltage sensor is not.

digitalentity, 2017-07-31:
@ThomasH3 you have GPS connected to UART1? This explains alot.

Built-in bootloader in DFU mode listens to activity on USB or UART1 (whichever happens first). GPS is sending data all the time, so bootloader switches to UART1 and listens to upload there.

ThomasH3, 2017-07-31:
I used TX/RX 7.

DzikuVx, 2017-09-26:
Flashed AnyFCF7 with 1.7.3 and 1.7.4 without any problems few days ago. Closing

closed 2017-09-26 by DzikuVx