iNavFlight/inav #1946: PID not using full servo throw on a flying wing

digitalentity, 2017-07-30:
With new servo mixer and default parameters roll/pitch PID is limited to 50% of max servo throw in ACRO/ANGLE/HORIZON on a flying wing. Reason unknown yet. Increasing PID gains doesn't help.

Settings: 1.7.2 @ defaults + FLYING_WING. `development` is affected as well.

@DzikuVx can you have a look?

DzikuVx, 2017-07-30:
This is how it should work if I understand the problem correctly. Correct
axis weight is 50% so one axis can output only half of servo throw. This is
why ppl were setting servo rate to 50

digitalentity, 2017-07-30:
So, not a bug, but a feature.

Dronek, 2017-08-01:
I also realized the reduced servo throw and had to increase servo rate from 50% to above 70% in order to increase the throw to a roughly similar amount as with 1.7.1

Do I understand correctly that 50% servo rate in 1.7.1 translates to 100% servo rate in 1.7.2 using flying wing mixer?

digitalentity, 2017-08-01:
@Dronek yes. You can use custom mixer to bring old behavior (with servo saturation) back. From today's flying I find new code more responsive at edge cases (i.e. commanding some elevafor when doing full stick roll)

DzikuVx, 2017-08-01:
Flying Wing mixer in 1.7.1 and all previous version was broken. One axis was able to completely saturate a servo without any room for second axis. 
Talking in `us` : with min and max at 1000 and 2000, when pitch requested 500 (max output) it was already reaching min or max. If roll was requesting something, where is was supposed to go? No room due to saturation.
Right now, there is no need to lower servo rate from 100% to 50% since fixed mixer does that automatically

DzikuVx, 2017-08-01:
@digitalentity lowering servo rate to 50% was a fix to broken mixer in < 1.7.2. 
Now in 1.7.2 , just by leaving servo rate at 100% you have the same behavior as previous releases  with servo rate 50%.

1.7.2 is not recommeneded to lower servo rate any more

DzikuVx, 2017-08-03:
So, if we already decided that this is how it was supposed to be working, I'm closing this

closed 2017-08-03 by DzikuVx