iNavFlight/inav #1950: GPS problem (rth and poshold iusse)

Lopu4991, 2017-07-30:
The GPS of my new first assembled quadcopter works fine: altitude, latitude and longitude are correct. The GPS I'm using is the Ublox neo-m8n from Banggood with the omnibus f4 pro v2. In poshold tha quad moves around. Also RTH mode are not working: when activated the quad suddenly drift away. I have never had the courage to see what it does; I immediatly switched mode. Althold is working fine. I have followed the GPS setup here https://github.com/iNavFlight/inav/wiki/GPS--and-Compass-setup. I have furthermore seen some videos on youtube.
Hope someone can help me

matoy, 2017-07-30:
hi,

i have exactly same symptoms; since i'm new to inav, i tested with 1.7.1

i have sp racing f3 deluxe (with baro and compass onboard) + 1 neo-m8n mini gps with UBLOX 3.01  and Galileo enabled.
accelerometer and compass are well calibrated, I read and triple checked the howto page on that subject.

pos alt: the quad stays at same altitude so it's okay
pos hold: toilet bowl effect or drone goes away.
rth doesn't work either but since pos hold doesn't work, i guess it's normal
in angle mode, the quad drifts a little bit but i could adjust that with "playing" with the board angle paramaters.

I know the compass is on board and not located in the gps and that is not recommanded but It seems to work and be well calibrated (i might have a difference of 5° at most to the reality with quad powered on).
gps works and fix sats.

I know about this issue:
https://github.com/iNavFlight/inav/issues/1525
but these CLI commands didn't help to solve the issue:
set gyro_notch1_hz = 200
set gyro_notch1_cutoff = 100
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 1
set dterm_notch_hz = 200
set dterm_notch_cutoff = 100

that could be great if you could help me solve this problem because i really would like to use mission waypoints features.

and thank you to all the team for all that great work.

please find a dump of my config below:
[INAV SPRACINGF3 1.7.1.txt](https://github.com/iNavFlight/inav/files/1186014/INAV.SPRACINGF3.1.7.1.txt)

digitalentity, 2017-07-31:
Please provide settings and logs.
Incorrect behavior in GPS-assisted modes is usually caused by these reasons:

1. Incorrect compass alignment and/or calibration - the most common problem
1. Bad GPS signal - too few satellites - a common problem with extremely cheap Chinese modules
1. Incorrect accelerometer calibration

matoy, 2017-08-07:
Hi, sorry for my lack of feedback but I had a major crash playing around due to a failing motor... My quad is in pieces and I'll have to wait for a new frame to arrive. I guess I'll build an Hexa with compass in gps this time and still try to plan missions :-)

DehKubbi, 2017-08-13:
I have a very similar experience also. I have the omnibus f4 from banggood as well and a hmc5883l plus a ublox neo6. I also isolated the mag with some steel that I grounded. The Mag still spits out data that does not make any sense at all. Sometimes it just sticks to some random value no matter which way you turn it. 
I got the mag working once, straight after calibrating it. After swapping the Battery this glitch appeared again. Could it be that the omnibus "forgets" it mag calibration and then uses wrong data?

I have recorded this issue with an dvr. The bigger tree is headed about 135°, yet the osd displays 260 and changes on rotation of the craft, but then goes back to 260. Interesting behavior indeed.
[https://youtu.be/Y930kVD8MwI](url)

digitalentity, 2017-08-14:
More likely a faulty compass chip. https://github.com/iNavFlight/inav/pull/1934 might help increase compass sensor reliability at expense of slower looptime.

VictorBucha, 2017-08-14:
Had just the same issue with neo8m. Observed all  the instructions,  followed all arrows. Set mag align @ 0 CV flip. BUT. The fact that you aligned arrows on FC and GPS means nothing: after setting CV 270 flip , my compass stated pointing at true North. Check everything!!!!

closed 2017-09-08 by digitalentity

grancarajo, 2017-10-08:
Similar case for my fixedwing, everything it's working fine horizon,angle,navlaunch, have at least 15 sat fix but can't activate any of gps mode. I be able to arm it but i can't activate rth or poshold.....hope you can help me......omnibus f3 with m8n gps from bangood

digitalentity, 2017-10-09:
@grancarajo RTH or poshold won't activate unless you are flying.

grancarajo, 2017-10-09:
Thanks for your reply.......!    so the only way to verify if it s working is flying...there is no other way to verify in gui if it's working?