iNavFlight/inav #1951: Sensor orientation

ShikOfTheRa, 2017-07-31:
For the sensor data graph in configurator, are X and Y for gyro, acc and mag supposed to match?

Like in original multiwii this is derived from (see “how should be the sensor axis directions”) linked here:
http://www.multiwii.com/wiki/index.php?title=Config.h#sensors_orientation

I was troubleshooting some odd behaviour with an FC and noticed it doesn’t match with what the target is configured as. RMDO clone in this case. 

I changed orientation so they match as below and not possible to fly. So I think maybe configurator is not right? 
TILT the MULTI to the RIGHT (left side up):
 - MAG_ROLL, ACC_ROLL and GYRO_ROLL goes up
TILT the MULTI forward (tail up):
 - MAG_PITCH, ACC_PITCH and GYRO_PITCH goes up

Ralf-W, 2017-08-01:
From my copter, that is flying correctly with GPS and MAG:
[image]
As long as the multi stays horizontal: 
acc_z=1, and acc_x=0, and acc_y=0.

Now TILT my MULTI to the RIGHT (left side up): 
acc_z=0, and acc_x=0, and acc_y=1.

This looks reasonable to me: 90° rotation around the x-axis makes acc_z change from 1g to 0g, acc_y change from 0g to 1g, and acc_x staying unchanged at 0g.

ShikOfTheRa, 2017-08-02:
Notice that whilst your acc "Y "goes +ve as you turn board to the right, it is the Gyro "X" value that increases. I would have expected ACC "and GYRO axis to match. 

Thanks for the info - it pretty much confirms what I worked out from physical test.

It’s probably just cosmetic, but confused the hell out of me when trying to set up a board from the sensor response on the graph.

Ralf-W, 2017-08-02:
There is a kind of match: Rotation around the x-axis makes the Gyro "X" detect and show a reading of some degrees/second around the x-axis.

closed 2017-09-08 by digitalentity