iNavFlight/inav #1957: Copter falling out of sky

stpfl00, 2017-07-31:
Board: SP EVO RACING
All started with a small crash on grass. i did not disconnect the battery. I rotated the copter in the right place because propellers an all the rest seemed to be fine. After takeoff and a 2 seconds hovering the copter just felled out of sky. the battery wasn't empty and the props were still rotating. So I disconnected the battery and 2 Hours later I gave it a new try. After hovering for about 30 seconds copter felled out of sky aggain. 
And now the strange thing kicks in. At the 3th and 4th Try Copter flew without any problems. I didn't changed anything. I now have one hour flight time after the strange behavior without any problem. 
I have no idea what it was or what the problem is. 
One thing I noticed is a humming sound from my matek 3.1 distribution board. But it's not really strong and I am not sure if it comes from the crash. 
 
I am very sad I had no SD card for logging the crash.

digitalentity, 2017-08-01:
@stpfl00 looks like a hardware issue. Humming sound coming from the BECs usually indicate malfunction or overload. Please check that nothing on your quad short-circuits.

DzikuVx, 2017-08-03:
Just like DigitalEntity wrote, this looks like a hardware issue and without blackbox log we are unable to identify the problem. I'm closing this for now, but if you would be able to record such a event in blackbox log, please share and reopen or create new issue

closed 2017-08-03 by DzikuVx