iNavFlight/inav #1959: Auto Tune "Tweaks"

moggiex, 2017-08-02:
Howdy,

Yesterday was the first time I have used the "Auto Tune" feature in iNav and frankly I found the expereince confusing due to the lack **any** confirmation of being in that flight mode or the saving of the settings afterwards.

With that in mind two suggestions to help improve the auto tune process for users:

**1. Show the mode on the OSD**
While I now know the auto tune mode isn't technically a mode, as far as the user is concerned it's on the modes tab and I expected the OSD to update the flight mode to show "TUNE" or something similar. 

Currently it stays showing ACRO or whatever mode iNav was in.

If "Auto tune" needs the aircraft to be in Acro mode, then can this just be enforced to make it easier for the user to not have another mode applied? 

For example I have several switches that have different flight modes attached to them, which could easily be on at the time. Thus imposing acro mode would reduce a mistake being made by the user. 

**2. Confirm the settings were saved**
After using the sticks to issue a save command changing out of the tune mode and after landing, there was no visual confirmation that the settings were saved.

It would have been much better for the OSD to have updated with the message "SAVED" in the middle of the screen, just like it does when you arm.

This is so that the user actually knows that the settings were actually saved.

Currently there is zero feedback on the OSD and it is assumed the action was made correctly.

Do those two adaptations sound reasonable to improve the auto tune process?

Matt

kabturek, 2017-08-10:
+1 for both
we need some more info about the modes that are on right now on the OSD. 
not sure if it should be displayed instead of the "main" mode, with  (ACRO(AT)) or there should be a secondary mode string in the OSD that can be displayed

Sedgee, 2017-08-16:
+1 for both
Would make life easier for sure as unless there is a laptop handy it's impossible to tell if any changes made.

fiam, 2017-09-24:
Part (1) done in https://github.com/iNavFlight/inav/pull/2201. Part (2) will come in a later PR once #2201 is merged.

fiam, 2017-09-26:
Moving to 1.9. Second part will be addressed by the upcoming message bus, which will allow us to shown messages in response to stick commands.

closed 2018-05-27 by stale[bot]