iNavFlight/inav #1962: Quad flips and crashes at random

mauimart, 2017-08-03:
I am struggling to get my quad to fly reliably using iNav on SPRacing EVO.  Things start great after initial liftoff and the quad hovers nicely in Horizon mode.  After a minute or two the quad will randomly flip and crash.  I have tried the same config on two different FC's (SPRacing EVO and Omnibus F3) with the same outcome.  I have attached the blackbox log of one of the crashes and the config dump.  Any ideas?

[LOG00003.TXT](https://github.com/iNavFlight/inav/files/1196359/LOG00003.TXT)
[iNav_config.txt](https://github.com/iNavFlight/inav/files/1196362/iNav_config.txt)

DzikuVx, 2017-08-03:
[image]

Your log clearly states that `motor 4` (front left) looses thrust. This is either motor or ESC or soldering problem. There is nothing a software can do in such a case. Recheck that motor/esc/cables and/or replace.

closed 2017-08-03 by DzikuVx

mauimart, 2017-08-03:
Thanks for the quick diagnosis.  I will investigate and swap out the ESC and fly again.  I have been having an issue with blackbox explorer.  The motor graph shows 100% for all motors throughout the entire flight even though in the table you can clearly see the raw values changing.  My decoded values are all 100%.  The screenshot you provided clearly shows the correct motor graph.  

[image]
[image]

JonasCz, 2017-09-11:
I had the same problem - eventually I found that I have to use the iNav BB explorer from [here](https://github.com/iNavFlight/blackbox-log-viewer) - the Betaflight one does not work correctly, even though it loads the file just fine and even changes to an iNav logo.