iNavFlight/inav #1963: We are running out of BOX IDs

DzikuVx, 2017-08-03:
With upcoming Runcam Split support we will run out of free box ids. Last 3 will be used by runcam and we are out.

Solutions:

1. Port array based masks: https://github.com/cleanflight/cleanflight/pull/2856 -> direct cherry pick  might break quite a lot of things
1. Bump `rcModeActivationMask` to `uint64_t` and extend MSP_STATUS_EX to send new mask along with old mask to keep compatibility with existing ground stations
1. Something else?

digitalentity, 2017-08-03:
I vote for `uint64_t` and modified API

oleost, 2017-08-03:
Keep compability with Cleanflight and Betaflight would probably be highly benifitial ? Even if it would mean more work as that will also help align with them?

digitalentity, 2017-08-03:
@oleost I'm not sure we should re-use CF/BF code here. 

While array-based approach is nice and scalable, it's probably an overkill - it extends usable box modes to potentially very large number, but do we need all of those? Simply extending to 64-bit will give us 35 more box modes to use - merely setting them up in Configurator would be a pain the back.

I'm appealing to YAGNI principle here - if we are not going to need more than 64 box modes in forseable future - we shouldn't waste time implementing support for them.

martinbudden, 2017-08-03:
I'd rather stay aligned with CF/BF unless there is a strong reason not to.

digitalentity, 2017-08-03:
@oleost @martinbudden actually you might be right. I forgot about 3rd party tools that talk MSP. There's also blackbox viewer which we'll probably want to keep generally compatible with INAV.

DzikuVx, 2017-08-03:
I would also mention, that there is a very important question: who is willing to port array solution and do Configurator changes from bit mask to array?
Do not count on me very much :)

digitalentity, 2017-08-26:
Fixed by #1966

closed 2017-08-26 by digitalentity