iNavFlight/inav #1969: Mag and Baro not supported in Inav but are in BetaFlight/CleanFlight

tonyintn, 2017-08-03:
Here seems to be new chips used in the latest SPRACING F3+ODS chips I have tried BetaFlight, and CleanFlight and they seem to be working but in Inav it does not seem to be covered.  
Things I have noted (chip set)
1 - L883/2046   is now M883/2023
2 - MPU 6050 no change
3 - STM32F303 CCT6 GH24Z 9U/449 is now GH24U 9u/GH718
4 - MAX 7456 / NBU 630 is now MAX7456 /MBU1703
5 - 25064FVsig /1619 is now 25064FVsig/1705

The Baro and Mag show up as Red in Inav I have tried Auto and each one they will not turn on
Beta Flight and Clean Flight they function fine 

I do have pic of the chip if that helps.
Thanks,

digitalentity, 2017-08-04:
The fact that sensors are detected in CF/BF doesn't mean that they function properly. INAV most likely detects them as well, but they fail to respond and are marked as "failing".

Please also see https://github.com/iNavFlight/inav/pull/1934, https://github.com/iNavFlight/inav/issues/724 and https://github.com/iNavFlight/inav/pull/1935.

tonyintn, 2017-08-04:
I did visually check the sensors on the testin page and they did seem to function in CF correctly 
This was a replacement chip I got from Banggood the fist one I got it did that behaviour so I figured it was defective I just received another today matching the same as the one couple weeks ago with the same behaviour with the Mag / and Baro. 
I was just wondering is they are perhaps using new chip sets that inav does not account for.

digitalentity, 2017-08-04:
Genuine HMC5883L is marked L883, genuine HMC5983 is marked A983. Your chip is probably a cloned one.

Please see this comment https://github.com/iNavFlight/inav/issues/724#issuecomment-319173886 - the guy also has the chip marked "M883" and it's not functional at 400kHz clock speed. At 200kHz it works fine.

tonyintn, 2017-08-04:
Awesome that worked, changing the clock speed from 400kHz to 200kHz did the trick sensors are alive and kicking.
Thank you, Thank you, Thank you!

closed 2017-08-05 by digitalentity