iNavFlight/inav #1973: VL53L0X rangefinder provides invalid readins above certain altitude

digitalentity, 2017-08-05:
Above certain altitude instead of reporting "out-of-range" VL53L0X rangefinder reports random altitude. Might be device initialization settings or simply a hardware limitation.

DzikuVx, 2017-08-05:
@digitalentity  `setSignalRateLimit(0.25f);` did solved the problem with bogus readout, but useful range dropped to less than a 1m. Solid readouts stopped at 60cm.

[LOG00001.TXT](https://github.com/iNavFlight/inav/files/1202572/LOG00001.TXT)

I will test other values.

On the other hand, I was able to do 10cm AGL flight over concrete...

DzikuVx, 2017-08-05:
`setSignalRateLimit(0.15f);` seems to be working too, no low altitude readouts. See the log. But range is still very low. @digitalentity I do not think 'long range' mode is indeed useful for VL53L0X. And this is evening over concrete, no direct sunlight

[LOG00001.TXT](https://github.com/iNavFlight/inav/files/1202591/LOG00001.TXT)

digitalentity, 2017-08-05:
In my tests bright light doesn't seem to affect performance too much. Still, I think VL53L0X might be useful indoors. as a part of hybrid optical/ultrasonic solution or for landing detection.

DzikuVx, 2017-08-06:
For a landing detector, yes. For terrain following only at very low altitude

digitalentity, 2017-08-06:
Closing this. Fix working. Sensor usefullness is another story :disappointed:

closed 2017-08-06 by digitalentity