iNavFlight/inav #1974: About TIM8 in drivers/timer_stm32F4xx.c

AkinakaTone, 2017-08-05:
At about line# 29, "#if !defined(STM32F411xE) && !defined(STM32F446xx)" is defined to exclude MPU who does not have TIM8. However, STM32F446xx does have Advanced TIM1&8 (Also by Reference Manual).
I noticed this, because I am occasionally testing Nucleo 303RE and 446RE for iNav with custom sensor board (heavy).

Since 446RE is not formally utilized in iNav, priority is low, but it is appreciated if revised at next convenient period. 

Thank you very much for your all the effots.

digitalentity, 2017-09-08:
No supported board uses STM32F446xx or STM32F411xE so this issue is irrelevant. @AkinakaTone please submit PR to clean up the code if you feel it's required. Thanks

closed 2017-09-08 by digitalentity

AkinakaTone, 2017-09-09:
Dear Sir

 

Thanks for your concern.

I totally agree with your final conclusion.

 

AkinakaTone