iNavFlight/inav #1975: ANYFCF7 LED strip not working in 1.7.2

flyfisher2, 2017-08-05:
LED strip worked in previous versions but not in 1.7.2

ahfynjj, 2017-08-18:
I have also encountered this problem

flyfisher2, 2017-08-18:
Thanks, that confirms that I don't have a hardware problem.

digitalentity, 2017-08-18:
@flyfisher2 @ahfynjj did you check latest defelopment firmware?

flyfisher2, 2017-08-19:
No, I only quickly tested the official release before flying to Scotland.
I will be away from home until the end of October, so unfortunately will not be able to do any more testing until then.

ahfynjj, 2017-08-19:
Latest development firmware? I do not know where to download？

digitalentity, 2017-08-20:
Try this one:
[inav_1.7.3_ANYFCF7.zip](https://github.com/iNavFlight/inav/files/1236967/inav_1.7.3_ANYFCF7.zip)

ahfynjj, 2017-08-20:
THS This firmware still can not drive the LED。
I tried the firmware of 1.7.0 can be used normally

sambas, 2017-08-21:
compare constants missing from HAL implementation.
    BIT_COMPARE_1 = 76; // timer compare value for logical 1
    BIT_COMPARE_0 = 38;  // timer compare value for logical 0

stronnag, 2017-08-21:
While we're here, is this significant (gcc 7 catch):
````
./src/main/drivers/light_ws2811strip.c: In function 'ws2811LedStripInit':
./src/main/drivers/light_ws2811strip.c:93:5: warning: 'memset' used with length equal to number of elements without multiplication by element size [-Wmemset-elt-size]
     memset(&ledStripDMABuffer, 0, WS2811_DMA_BUFFER_SIZE);
     ^~~~~~
````
ledStripDMABuffer being uint32_t (vice uint8_t) on F4 / F7 ?

capblack, 2017-09-01:
Hi , also Motor #9 output as buzzer output , is not work

digitalentity, 2017-09-08:
@sambas can you verify if this is still relevant?

sambas, 2017-09-11:
fixed in https://github.com/iNavFlight/inav/pull/2057

closed 2017-09-11 by sambas