iNavFlight/inav #1985: Fixedwing throttle control in various flight modes.

oleost, 2017-08-09:
Think it's needed to rethink how the throttle stick should act in various flight modes, with and without airspeed sensor.

Specific:

1. Althold
2. Poshold
3. Poshold and althold combined

My thoughts so far

New variables named nav_fw_speed_mode which can be gps_speed or air_speed

Althold:
0-95% controls speed it controls scales between new variables nav_fw_minspeed and nav_fw_maxspeed. 
Above 95% it override speed controller and give 100% throttle. Minimum throttle will still be constrained by "nav_fw_min_thr"

SoulPictures, 2017-08-09:
Sounds like an idea, but should be usable as a MODE due to possibilities of not noticing dirt or other stuff within the pitot tube... ? Maybe as an Mode with configurable Speed as 50% Throttle? so you could set the best cruise speed for your model as 50% throttle and scale up and down from there through stick movement ?

fiam, 2017-10-07:
@oleost I think it would be better to put that on a POT, to avoid accidental triggers. Also, I think is better to leave the THR stick alone while in NAV modes, so it will be on a position that will make the aircraft fly when you switch to manual control again.

closed 2018-05-27 by stale[bot]