iNavFlight/inav #1986: Boards hangs when trying  to configure "Modes" when GPS is configured and not present.

oleost, 2017-08-09:
Spracingf3mini
INAV 1.7.3 development branch.
Configurator 1.7.2

Turn on GPS on an uart, enabled GPS in configurator, then try and enter "Modes" tab.

This seems to hang the board and a power cycle is requirred,

DzikuVx, 2017-08-09:
Looks like bug lies in MSP_BOXNAMES frame somewhere

digitalentity, 2017-08-10:
@oleost can you provide a complete dump of failing build? I can't reproduce the issue on other boards.

oleost, 2017-08-10:
@digitalentity Full CLI dump https://pastebin.com/G6KBLnnn

With that dump go to Configuration, enable gps and try to enter "Modes" tab

But I take back that the board locks up, if I press dicsconnect, and then connect again it will connect. ( But still get stuck on "Waiting for data ..." if I try and enter Modes tab

Disable GPS again and I get enter modes tab.

DzikuVx, 2017-08-10:
@digitalentity I was able to reproduce on AirbotF4. If GPS is enabled but not connected, this is what is happening during access to `MSP_BOXNAMES` or previous frame in the queue, I was unable to determine

digitalentity, 2017-08-10:
Ok. I know where the problem lurks. With enabled GPS `MSP_BOXNAMES` response length in this setup is 283 bytes which exceeds max MSP payload length of 255 bytes. FC inserts jumbo frame header to transfer longer payload, but Configurator apparently doesn't support it.

Intermediate solution - make BOX names shorter. Ultimate solution - correctly decode JUMBO frames in Configurator. I believe this should be discussed along with totally new MSP packet format as mentioned by @stronnag here https://github.com/iNavFlight/inav/pull/1976#issuecomment-321307320

DzikuVx, 2017-08-10:
@digitalentity adding Jumbo frames to Configurator should be few hours job. And I really really wanted to do it for a long time now. But since I'm leaving tomorrow for 2 weeks, it will not happen very soon. 
Let's say, I can do it as the next think. Until then, we either put it on hold or make names shorter

digitalentity, 2017-08-10:
I'm not sure we need jumbo frames. See my comment here https://github.com/iNavFlight/inav/pull/1976#issuecomment-321497036

DzikuVx, 2017-08-10:
I'm not sure here. The thing is that both firmware and GCS would have to learn to use new protocol.

digitalentity, 2017-08-10:
Yes, but we'll be doing MSPv2 anyway, why bother with jumbo frames?

DzikuVx, 2017-08-10:
For BB logs download

digitalentity, 2017-08-10:
BB logs download can use MSPv2 natively in Configurator

closed 2017-08-27 by digitalentity