iNavFlight/inav #1990: Smartport telemetry on SS in SPRacing F3

teckel12, 2017-08-11:
I'm not sure #1932 is resolved. I have the same situation and I've redone the connection a couple times. I'm also using SP Racing F3 (non-EVO) with the X4R-SB Rx and using soft serial connection with pins 5 & 6 tied together and going to the S.Port connector on the Rx.

It won't work at all, then works a few times in a row, then doesn't work at all again. If I can get it running and get a GPS fix quickly enough and launch before telemetry drops, I'll get the telemetry for the entire flight without a problem. But, if it takes a while to get a GPS lock, in almost every case telemetry will drop after a few seconds before arming.

I've resorted to the following procedure, connect a bench battery, power up, wait for GPS fix (the telemetry will drop after a few seconds). I then disconnect the bench battery and connect a flight battery. Because it previously got a GPS fix, it will typically get a GPS lock quick enough for me to launch and get telemetry for the entire flight. But, this is still hit or miss.

Something just isn't right and I'm not sure what the problem is. I may try to connect pins 5&6 to a serial port and read them on the PC to maybe find out what is happening. But I'm at a total loss as everything is setup correctly (and working for entire flights when it does work).  I did read somewhere that telemetry will stop if you don't arm, but it sounds like this was old information (and probably Cleanflight as well).

UPDATE:  I did the following today.  I `set nav_extra_arming_safety = OFF`, connected the battery, and immediately armed without a GPS lock.  I then waited for the GPS to lock and flew as normal, without telemetry ever dropping.  I tried this with a few batteries without issue.  This was the first time I was able to burn through multiple batteries and have telemetry the entire time.  I like the `nav_extra_arming_safety` setting and I really don't want to drain my battery spinning the props while waiting for a GPS fix.  But, this is the only solution till this issue can be resolved.

teckel12, 2017-08-11:
`inav/src/main/telemetry/smartport.c` line: 127

```
#define SMARTPORT_SERVICE_TIMEOUT_MS 1 // max allowed time to find a value to send
```

It appears the telemetry identifier will be closed if there's no value to send in 1ms.  I didn't mention in my original message that not all of the telemetry identifiers will drop all the time.  Sometimes, just some of them drop.  Or, they slowly drop every couple seconds till all are dropped.  I lowered my loop time to reduce CPU usage thinking that may help, but it didn't.  Maybe 1ms is just quite long enough of a timeout?

I'm not sure why I don't try this on my own and compile my own iNav, being a programmer for a living.  I'll work on this over the weekend and report back if I find anything.

teckel12, 2017-08-11:
I'll try installing a diode between 5 & 6 tonight to see if that changes anything.  I assume the diode would be installed like the following as the purpose is to keep the port open by receiving signals on the RX port:

TX(6) --▷|-- RX(5)

Then, wire TX(6) to the S.Port of the Rx.

VictorBucha, 2017-08-15:
I confirm same situation, but I'm using LTM telemetry via Bluetooth. EZ-Gui sometimes reports "no data received". It is probably due to CPU load of about 40% after reducing freq on i2c bus.  I'm no good in recompiling firmware to increase SMARTPORT_SERVICE_TIMEOUT_MS 1
can it be done via CLI?

teckel12, 2017-08-15:
@VictorBucha I've not only tried increasing the timeouts (there's actually another as well) but I've even totally removed the timeouts and telemetry sensors are still dropping.  I've lowered my CPU load without noticing a difference in the drop-out rate.  I'm still working on this, and I'll be trying a few things tonight as well that I've thought of that could be causing it.

Did you have a problem with previous versions of iNAV?  I only tried telemetry starting with v1.7.1 so I'm unsure if the problem always existed or just started.

VictorBucha, 2017-08-16:
@teckel12 in 1.7.1 everything worked smoothly. my 1.7.1 drone crashed so cannot upgrade and test it.  I have no Ublox7 to test as it seems ublox8 caused much trouble.

teckel12, 2017-08-16:
Pick-up from closed issue #1932 

@digitalentity @Zrelos I created a pull request #2002 which looks to address the problem discussed here as well as enhance/cleanup/fix/document a few SmartPort telemetry items.

My testing:  I tried drastically slower loop times and disabling features but telemetry would still drop.  I also tried disabling sensors, disabling timers, etc. and telemetry would still drop.  CPU load was very low (3%) and running an F3 target.  Pull #2002 does appear to make the issue a little better by increasing `SMARTPORT_SERVICE_TIMEOUT_MS`.  With pull #2002 I was able to set `set nav_extra_arming_safety` back to `ON` and didn't need to use a bench battery to get a GPS lock before switching to my flight battery.  Telemetry sensors would eventually drop, but it was more like 5 minutes instead of 7 seconds that I experienced previously.

@digitalentity I'll look at the SoftSerial code to see if I can track down the problem.  SoftSerial even works on ATmega328 microcontrollers, so my guess is that there's an issue with the code instead of an F3 running at 3% load isn't fast enough for 57600 baud.  But, #2002 does make SmartPort telemetry usable again when using SoftSerial (and adds a few new features).

BbypK, 2017-08-16:
Tried a custom SPRacingF3 Evo build with only one timeout change.
https://github.com/iNavFlight/inav/blob/81b85dec1236bf829c6079578f81fbb0410c49a3/src/main/telemetry/smartport.c#L130
From 7000 to 70000.

Using an XSR Receiver, I lost telemetry in some seconds after battery plugged, but restored after gps lock and / or craft armed. Only tried on ground so i dont know yet if telemetry works without drops in flight.

teckel12, 2017-08-16:
@BbypK I tried totally removing the `SMARTPORT_NOT_CONNECTED_TIMEOUT_MS` logic from the code and telemetry sensors were still lost.  One strange thing about this problem is that once in the air I never lose a sensor, not once.  It's only before arming (and sometimes within seconds).

I'm going to dive deeper into this problem as it seems it could be a softserial problem.  My guess is that it has to do with wiring the RX & TX together to keep the softserial port open as that isn't required when using a physical port.  Also, everyone who's having a problem with dropped sensors is running via softserial.

BbypK, 2017-08-16:
Physical port can use a single pin in bidirectional mode, softserial pins are unidirectional so TX and RX have to be joined together.

FrSky telemetry is hardcoded at 57600 baudrate and physical port can handle that speed natively, softserial depend on cpu and without compromise a lot of load is limited at 19200 baudrate as far as i know.

teckel12, 2017-08-17:
@BbypK I'm not convinced it's just a SS limitation for two reasons:

- First, a 8-bit Arduino ATmega328 can do SS, even multiple SS ports running at 115200 baud.  And even when I select the lowest loop time and turn off as many options as possible (even removing them from the compile) it still will drop sensors.
- Secondly, once armed sensors NEVER drop.  I only ever get sensors to drop when I'm searching for GPS and not armed.  A "solution" is to `set nav_extra_arming_safety = OFF` and arm right away while waiting for GPS, then launch.  When I've done this, I've NEVER had a sensor drop.

For these reasons, I just can't accept that the F3 isn't fast enough to communicate at 57600 baud when my CPU load is at only 3%.  It could be a bug in the SS code (there was some fussing with it recently and specifically to problems with the SPRACINGF3 target).  It could also be something else that would effect SmartPort or SS before being armed.

BbypK, 2017-08-17:
@teckel12 Don't get me wrong, I'm sure that you know this better than me. 

My board is an F3 Evo, so it isn't only related to SPRACINGF3 target. But I have to try to backup my config and flash the board with cleanflight and test telemetry to see if its Inav code related or faulty board.

teckel12, 2017-08-17:
@BbypK Cleanflight uses significantly different smart serial code than iNav.  However, the difference between SmartPort code is slight, mainly iNav adds the not connected timeout (which doesn't appear to be the problem).  Maybe I'll try to fold the Cleanflight SmartPort code back into iNav and see what that does.

Another possible "solution" for me is to swap GPS and SmartPort.  Moving GPS to the soft serial port (which runs at 38400 baud) and SmartPort to a physical serial port.  It's very possible that the X4R-SB receiver isn't very tolerant of slight timing issues inherent with using smart serial.

If this is the "solution" it's not all bad.  It gives a clear solution (switch ports) and by me poking around I added some new features to SmartPort.

On that note, is there anything you would like to see added to SmartPort telemetry?  It can be expanded if there's any requests.  I was thinking of a "RTH suggestion telemetry" that's based on battery capacity used, battery capacity, distance from home, height, RTH method, climb/travel/descend speed.  Basically, the FC would decide if you'll run out of battery before flying home based upon real RTH values.  If so, it would send telemetry to your transmitter indicating that you should really fly home.  You can then make your transmitter set off an alarm, give a warning message, or even trigger the RTH switch automatically via a special function.

BbypK, 2017-08-17:
@teckel12 Yes, I saw the diferences in the code and also, in Cleanflight, isn't necessary the use of a diode in the SP Evo using softserial, so i want to see if the telemetry drop affects only to Inav code or hardware has some relation to it.

I use a Naza GPS unit that runs on 115200 bauds ( using less speed cause mag not detected a lot of times ) so running it from SS it's not an option.
I could try with osd at same baudrate that Smartport the next week if nobody can do some test.

teckel12, 2017-08-18:
@BbypK I'll be doing some testing tonight and over the weekend with this.  The part that REALLY drives me up the wall is how telemetry never drops once armed.  I need to confirm this, running through a few batteries to be sure, as this could allow me to focus in on a solution.

teckel12, 2017-08-22:
I've got a really stable iNav branch which all indicators tell me it will be merged into iNav.  Telemetry is now rock-solid over soft serial, with one exception... heat.

The thing that's been bothering me forever with this problem is how telemetry would never drop when armed.  Come to find out, the reason was the heat generated from my VTx.  It would heat up and cause telemetry to fail.  When armed, air-cooling prevented over-heating and sensors were never lost.  While waiting for a GPS fix, it could heat up and sensors would fail in the process.

I can literally blow on my quadcopter without touching it and telemetry will start again. Which is why it never drops when armed, because the blades are cooling the quadcopter. If I turn off my VTx the problem also goes away as the VTx generates most of the heat.

In any case, my iNav code changes do help with telemetry sensors lost.  And in the process I've added a few features and a foundation that will allow for telemetry expansion (like adjusting PIDs from your Tx).

digitalentity, 2017-09-08:
@teckel12 please remind me what this issue is tracking? Your Sport changes are merged to development and will soon make it to master. Is this still relevant?

teckel12, 2017-10-01:
@digitalentity This issue can be closed as a result of 1.7.3

closed 2017-10-01 by digitalentity