iNavFlight/inav #1994: OLED display on F1 board

dispetto77, 2017-08-12:
Hi. I was trying to compile iNav for fixed wing on a illuminati32 board.
I would like to turn on the OLED display . I have enough space because I used pwm and I disabled all the hardware and features I don't need.
How could I Add Display Feature in Naze fw? I tried several ways without success. Thank you

fiam, 2017-08-13:
@dispetto77 You'll want to add USE_DASHBOARD to your build config. I did a quick and dirty test and it seems the easier way would be to add it to src/main/target/common.h, just after #pragma once:

```diff
diff --git a/src/main/target/common.h b/src/main/target/common.h
index 08faf3d8..05ad03b7 100755
--- a/src/main/target/common.h
+++ b/src/main/target/common.h
@@ -16,6 +16,7 @@
  */
 
 #pragma once
+#define USE_DASHBOARD
 
 #define I2C1_OVERCLOCK false
 #define I2C2_OVERCLOCK false
```

Then just type:

```sh
make NAZE
```

Note that it currently overflows the target by 2774 bytes, so you should have to disable a few other things in order to make it work.

dispetto77, 2017-08-13:
Thank you. I did it without success in the past days. Maybe my OLED is broken if the software part is ok.... Thank you very much

giacomo892, 2017-08-14:
Not all OLED displays are compatible with iNAV

dispetto77, 2017-08-14:
Yes I know that

dispetto77, 2017-08-14:
Hi
I compiled using the option wrote up but it didn't work. Feature dashboard is on but the oled is off. I used a CRIUS OLED with 14pin mod and this banggood OLED
https://m.banggood.com/Cleanflight-Firmware-OLED-Display-Flight-Controller-Status-Displayer-for-NAZE32-CC3D-p-1058086.html?utm_source=google&utm_medium=cpc_ods&utm_content=jeff&utm_campaign=Smlr-ds-naze32&gclid=EAIaIQobChMIjNyCn4zX1QIV6hbTCh1OcgjREAAYAiAAEgLv5vD_BwE

DzikuVx, 2017-08-16:
There indeed might be something wrong with OLED code since I failed to connect it to AnyFC F7 as well, but never really put any effort into it since I was planning to get rid of OLED either way

Od: dispetto77
Wysłano: poniedziałek, 14 sierpnia 2017 20:03
Do: iNavFlight/inav
DW: Subscribed
Temat: Re: [iNavFlight/inav] OLED display on F1 board (#1994)

Hi
I compiled using the option wrote up but it didn't work. Feature dashboard is on but the oled is off. I used a CRIUS OLED with 14pin mod and this banggood OLED
https://m.banggood.com/Cleanflight-Firmware-OLED-Display-Flight-Controller-Status-Displayer-for-NAZE32-CC3D-p-1058086.html?utm_source=google&utm_medium=cpc_ods&utm_content=jeff&utm_campaign=Smlr-ds-naze32&gclid=EAIaIQobChMIjNyCn4zX1QIV6hbTCh1OcgjREAAYAiAAEgLv5vD_BwE

closed 2018-01-20 by DzikuVx