iNavFlight/inav #1996: RTH Prevents Arming

mgrimes1907, 2017-08-13:
I am using an omnibus F4 Pro V3 on a flying wing. I have tried the code for the Omnibus F4 V3 and what I found was that the SoftSerial port would not work on that program. So I flashed the controller with the Omnibus Pro and found that I had SoftSerial ports that work. While setting the rest of the settings, everything went fine till I set my Modes. All the modes work fine untill I select "NAV RTH" on AUX 1.

After selecting "NAV RTH" it will not arm. I know this function was working when flashed with
OMNIBUS F4 V3 code. So I thought to myself, what if I set it up under "FAILSAFE". Same result, won'tarm. Go back and set "FAILSAFE: to "DO NOTHING" and it will arm. 

Is there a work around? If I set set the "nav_extra_arming_safety" to off, will the "NAV RTH" function work?

Any help would be great.

DzikuVx, 2017-08-13:
it is a correct behavior. RTH returns to place where you armed so where you
want it to return? Somewhere else?

mgrimes1907, 2017-08-15:
I am sorry, I wasn't clear on the issue. I theinave configurator ,  while setting up modes, everything works great until I set the channel and the range for that channel to "NAV RTH". After that, I can't arm my unless I change the extra safety arming to no in the CLI command

giacomo892, 2017-08-15:
Do you have a GPS fix?

oleost, 2017-08-15:
To answer your question:

nav_extra_arming_safety is there to ensure you have an valid GPS fix before arming.

When turned on it prevent arming when not having a valid fix.

When turned off it will allow to arm. If you decide to arm before an valid fix is optained RTH will not work. If you wait untill you have an valid fix RTH will work as usual.

Now closing this issue. Use rcgroups for any more support.

closed 2017-08-15 by oleost