iNavFlight/inav #1997: Generated CLI settings (discussion)

fiam, 2017-08-13:
I've been doing some experiments with cli.c in order to save some space without sacrificing functionality and it seems that introducing generated CLI settings (namely clivalue_t valueTable[] in cli.c) from a description would provide space savings at least around ~1K for all targets. The only disadvantage would be that developers would need to regenerate the generated file(s) when introducing or modifying settings and might need additional tools installed in their dev environment.

My prototype implementation currently uses YAML to describe the settings and a Go program to process them and generate a .c and a .h file which are then included by cli.c

Space savings currently come from two sources:

- Setting name compacting: setting names are split into words which are stored in an array. Then each setting name consists of a series of variable length encoded unsigned ints (MSB high indicates more bytes for the number follow). This requires 6 bytes per setting name and saves between ~700 and ~1300 bytes per target (targets with more settings get more savings, since there's some overhead introduced by the functions for checking and comparing setting names).
- Moving constants into an array: Currently each setting requires 32 bits per entry to store the min/max values. However, there are only 105 different values. Storing them in an array and using 14 bits to index min and max into the array saves ~300K. This could probably be improved by using different value arrays (one for uint8_t, one for float, etc...), since the current implementation only uses one int32_t array for storing all values.

I'd like to know the opinion of the core developers regarding this change before I spend more time on it. Do you think it is worth to introduce this complexity in exchange for the additional flash space it provides?

digitalentity, 2017-08-14:
This is really interesting. Can you show us some generated code that's used in cli.c?

fiam, 2017-08-14:
@digitalentity I've just pushed the branch to my fork, you can check it here: https://github.com/fiam/inav/tree/cli

The changes are in fc/settings.yaml, fc/settings.{h,c}, fc/settings_generated.{h,c}, fc/settings.go and fc/cli.c.

Note that the current implementation doesn't quite work. The FC will hang during a dump or a diff, but I haven't had time to fix it yet. Also, there's a bit of low hanging fruit in the functions in settings.c, since currently they just copy the whole setting name to a buffer and then call into strcmp, strstr... etc... This will eventually be replaced by functions that don't need to reconstruct the name.

Also, the current implementation generates files which work for all targets and are intended to be committed to the repo. This results in some words stored in the word array that aren't really used by some targets. A better strategy would running the settings generator at compile time and generating files specific for the current target/CFLAGS. This would allow quite a few more savings, like generate less than 256 words in F1 targets allowing direct byte addressing for the words array and getting rid of the 16 bits in the pgn_t member (we could use a table with the each group starting index).

As you can see, there's still quite a long way to go until this could get merged, but I guess we need to start somewhere :-).

fiam, 2017-08-18:
@digitalentity I've just pushed some updates and the code is now working on my tests on F3 and F4. I don't have any F1 boards right now, so it would be helpful if someone could give it a try.

Currently, the generator is written in Go, so you'll need to install it in your system as well as go getting the package for the yaml parser:

```sh
go get -v gopkg.in/yaml.v2
```

Eventually, I'll probably rewrite the generator in Ruby, since it seems the default installation on macOS includes support for yaml parsing and installing it on Linux or Windows should be easy enough.

Running make SOME_TARGET should automatically generate the settings files, but the Makefile still needs some tweaks to avoid rebuilding the settings every time.

Right now, value packing (the .minmax and .max fields) is not yet properly implemented, so I haven't committed those changes yet. For now, the saving are around 2.5K on F1 targets, which allows building on NAZE again.

digitalentity, 2017-08-18:
@fiam great! I'll install Go on my system and try it. I think script itself should be moved to `/tools` folder (a new one alongside `src` and `docs`

martinbudden, 2017-08-19:
Good stuff. Not a fan of Ruby, though. It's a horrible language. Would rather keep Go or use python.

stronnag, 2017-08-19:
Holy Wars!!!!! Ruby is wonderful. We already have a ruby tool in the configurator project (that generates the model images).

digitalentity, 2017-08-19:
TBH, I'm not fond of either language :rofl:

fiam, 2017-08-19:
Just to clarify, I mentioned Ruby as the most likely candidate not because I like it (in fact, it's pretty low in my list of go-to languages), but because it's the only alternative installed with YAML support  by default on macOS. Given that the settings generator will run at build time, any tools required to run it become required tools to build INAV and I'd rather not make it difficult for end users to compile their own builds.

If we can all agree to use other format to write the settings, we can explore other languages. Being installed by default on macOS is a huge plus, but I'm open to pretty much anything. At this point the generator is just a few hundred lines so rewriting it is not much of a hassle, but I'd like to get this settled before I add more functionality to it.

So, from my PoV I see the following alternatives as the best ones:

* YAML for the settings file, Ruby for the generator: Not that I like Ruby too much, but the tradeoff seems convenient in order to be able to use YAML to write the settings.
* JSON for the settings file, Python for the generator: I do prefer Python to Ruby, but JSON is a pain to write manually. Also, there's the Py2 vs Py3 issue and we'll probably need to eventually support running the generator on both of them, which will introduce additional maintenance and testing burden.
* Keeping the generator in Go with YAML settings: This has the disadvantage of requiring users to install Go, set it up properly and install a 3rd party package, which is probably a bit too much. The advantages of this of this approach include that Go is pretty easy to learn and use and the generator is already working and functional if we choose to keep it this way. I chose it for the prototype because I'm very comfortable in Go so it allows me to iterate fast. Also, at the start it wasn't clear if we could get by just generating files valid for all targets/configurations which would be committed to the repo (thus, imposing the burden of having the right tools just on the developers, not on end users), while now it's clear that generating per-build files yields way better results (specially on smaller targets).

I'm intentionally leaving out other simpler and less structured formats like CSV, since writing the settings in those would be a real pain and very error prone.

stronnag, 2017-08-19:
Really don't care about the language; I'd have favoured JSON over YAML, but that's also a 'so what'.
Anyway, just flashed the MATEK F405 with the CLI  fork, restored the previous `diff`, saved the new `diff` -- no difference between them. Let's fly it, if it doesn't rain ...

martinbudden, 2017-08-19:
Sorry, I did not mean to start a language war. I don't have a strong objection to using any particular languate.

I don't thing we should users of a particular OS though (and I say this as a Mac user). So if the choice is Ruby it should be for reasons other than it is installed by default on OSX.

stronnag, 2017-08-19:
Nor I. Anyway, flight tested. No problems (so a second successful test of #1966). Nice space saving.

fiam, 2017-08-20:
@stronnag Thanks for flight testing it, I really appreciate that.

I've just pushed some more updates to my branch, implementing very simple value packing/indexing for min and max setting values. This brings the space saving almost to 3K on F1 targets.

```
| Filename              | Prev size | Cur size | Change |
---------------------------------------------------------
|         inav_CC3D.elf |    125264 |   122276 |  -2988 |
| inav_OMNIBUSF4PRO.elf |    205262 |   201910 |  -3352 |
|   inav_SPRACINGF3.elf |    180948 |   177836 |  -3112 |
```

digitalentity, 2017-08-20:
@fiam I really like to get this merged as soon as possible. Can you create a PR and update the build requirements in the docs?

stronnag, 2017-08-20:
Flew yesterday's version on my tricopter today (for something slightly different). No problems.

stronnag, 2017-08-20:
Currently make is not working:
````
commit 89eb050be31d46163b87cdb069bde0ed950b4698 (HEAD -> cli, origin/cli)
Author: Alberto García Hierro <alberto@garciahierro.com>
Date:   Sun Aug 20 13:34:38 2017 +0200

    Initial attempt at encoding constants used for min/max settings
    
    Pretty naive approach for now. Saves ~400 bytes on F1 targets.
````
Thusly
````
$ make V=1
rm -f ./src/main/fc/settings_generated.h ./src/main/fc/settings_generated.c
CFLAGS="-mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Wdouble-promotion -flto -fuse-linker-plugin -Os  -I./src/main -I./src/main/target -I./lib/main/STM32F30x_StdPeriph_Driver/inc -I./lib/main/CMSIS/CM1/CoreSupport -I./lib/main/CMSIS/CM1/DeviceSupport/ST/STM32F30x -I./lib/main/MAVLink -I./src/main/target/SPRACINGF3 -ggdb3 -DDEBUG -std=gnu99 -Wall -Wextra -Wunsafe-loop-optimizations -Wdouble-promotion -ffunction-sections -fdata-sections -DSTM32F303xC -DSTM32F303 -DFLASH_SIZE=256 -DHSE_VALUE=8000000 -DUSE_STDPERIPH_DRIVER -DSPRACINGF3 -D'__FORKNAME__="inav"' -D'__TARGET__="SPRACINGF3"' -D'__REVISION__="89eb050b"' -save-temps=obj -MMD -MP" ./src/main/fc/settings_gen . ./src/main/fc/settings.yaml
270 settings
word table has 252 entries
each setting name uses 5 bytes
3021 bytes estimated for setting name storage
mkdir -p obj/main/SPRACINGF3/fc/
echo %% cli.c ""
%% cli.c 
arm-none-eabi-gcc -c -o obj/main/SPRACINGF3/fc/cli.o -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Wdouble-promotion -flto -fuse-linker-plugin -Os  -I./src/main -I./src/main/target -I./lib/main/STM32F30x_StdPeriph_Driver/inc -I./lib/main/CMSIS/CM1/CoreSupport -I./lib/main/CMSIS/CM1/DeviceSupport/ST/STM32F30x -I./lib/main/MAVLink -I./src/main/target/SPRACINGF3 -ggdb3 -DDEBUG -std=gnu99 -Wall -Wextra -Wunsafe-loop-optimizations -Wdouble-promotion -ffunction-sections -fdata-sections -DSTM32F303xC -DSTM32F303 -DFLASH_SIZE=256 -DHSE_VALUE=8000000 -DUSE_STDPERIPH_DRIVER -DSPRACINGF3 -D'__FORKNAME__="inav"' -D'__TARGET__="SPRACINGF3"' -D'__REVISION__="89eb050b"' -save-temps=obj -MMD -MP fc/cli.c
arm-none-eabi-gcc: error: fc/cli.c: No such file or directory
arm-none-eabi-gcc: fatal error: no input files
compilation terminated.
make: *** [Makefile:955: obj/main/SPRACINGF3/fc/cli.o] Error 1
````

fiam, 2017-08-20:
@stronnag Thanks for the heads up. Seems to work fine with GNU make 3.81. Could you please let me know your make variant and version so I can try to reproduce it?

stronnag, 2017-08-20:
It was fine yesterday!

````
$ make -v
GNU Make 4.2.1
````

stronnag, 2017-08-20:
And you really don't want to know what happens with arm-none-eabi-gcc 7.1 (or specifically arm-none-eabi-g++ 7.1). Just so we appreciate what's coming.
````
[jrh@tigger inav-fcgen]$ make
/tmp/667876137/test.cpp: In instantiation of 'class Fail<0>':
/tmp/667876137/test.cpp:38:32:   required from 'class expr_VBAT_SCALE_MIN<0>'
/tmp/667876137/test.cpp:39:37:   required from here
/tmp/667876137/test.cpp:33:6: error: static assertion failed: FAIL
      static_assert(V == 42 && 0 == 1, "FAIL");
      ^~~~~~~~~~~~~
/tmp/667876137/test.cpp: In instantiation of 'class Fail<255>':
/tmp/667876137/test.cpp:40:28:   required from 'class expr_VBAT_SCALE_MAX<255>'
/tmp/667876137/test.cpp:41:37:   required from here
/tmp/667876137/test.cpp:33:6: error: static assertion failed: FAIL
/tmp/667876137/test.cpp: In instantiation of 'class Fail<100>':
/tmp/667876137/test.cpp:42:28:   required from 'class expr_FLAPERON_THROW_MIN<100>'
/tmp/667876137/test.cpp:43:45:   required from here
/tmp/667876137/test.cpp:33:6: error: static assertion failed: FAIL
/tmp/667876137/test.cpp: In instantiation of 'class Fail<500>':
/tmp/667876137/test.cpp:44:28:   required from 'class expr_FW_ITERM_THROW_LIMIT_MAX<500>'
/tmp/667876137/test.cpp:45:57:   required from here
/tmp/667876137/test.cpp:33:6: error: static assertion failed: FAIL
/tmp/667876137/test.cpp: In instantiation of 'class Fail<-32768>':
/tmp/667876137/test.cpp:46:28:   required from 'class expr_INT16_MIN<-32768>'
/tmp/667876137/test.cpp:47:27:   required from here
/tmp/667876137/test.cpp:33:6: error: static assertion failed: FAIL
/tmp/667876137/test.cpp: In instantiation of 'class Fail<1>':
/tmp/667876137/test.cpp:48:28:   required from 'class expr_RSSI_SCALE_MIN<1>'
/tmp/667876137/test.cpp:49:37:   required from here
/tmp/667876137/test.cpp:33:6: error: static assertion failed: FAIL
/tmp/667876137/test.cpp: In instantiation of 'class Fail<2000>':
/tmp/667876137/test.cpp:50:28:   required from 'class expr_PWM_RANGE_MAX<2000>'
/tmp/667876137/test.cpp:51:35:   required from here
/tmp/667876137/test.cpp:33:6: error: static assertion failed: FAIL
/tmp/667876137/test.cpp: In instantiation of 'class Fail<6>':
/tmp/667876137/test.cpp:52:28:   required from 'class expr_CONTROL_RATE_CONFIG_ROLL_PITCH_RATE_MIN<6>'
/tmp/667876137/test.cpp:53:87:   required from here
/tmp/667876137/test.cpp:33:6: error: static assertion failed: FAIL
/tmp/667876137/test.cpp: In instantiation of 'class Fail<1000>':
/tmp/667876137/test.cpp:54:28:   required from 'class expr_ACC_TASK_FREQUENCY_MAX<1000>'
/tmp/667876137/test.cpp:55:53:   required from here
/tmp/667876137/test.cpp:33:6: error: static assertion failed: FAIL
/tmp/667876137/test.cpp: In instantiation of 'class Fail<180>':
/tmp/667876137/test.cpp:60:28:   required from 'class expr_CONTROL_RATE_CONFIG_ROLL_PITCH_RATE_MAX<180>'
/tmp/667876137/test.cpp:61:87:   required from here
/tmp/667876137/test.cpp:33:6: error: static assertion failed: FAIL
/tmp/667876137/test.cpp: In instantiation of 'class Fail<10>':
/tmp/667876137/test.cpp:66:28:   required from 'class expr_HEADING_HOLD_RATE_LIMIT_MIN<10>'
/tmp/667876137/test.cpp:67:63:   required from here
/tmp/667876137/test.cpp:33:6: error: static assertion failed: FAIL
/tmp/667876137/test.cpp: In instantiation of 'class Fail<750>':
/tmp/667876137/test.cpp:70:28:   required from 'class expr_PWM_PULSE_MIN<750>'
/tmp/667876137/test.cpp:71:35:   required from here
/tmp/667876137/test.cpp:33:6: error: static assertion failed: FAIL
/tmp/667876137/test.cpp: In instantiation of 'class Fail<80>':
/tmp/667876137/test.cpp:74:28:   required from 'class expr_YAW_JUMP_PREVENTION_LIMIT_LOW<80>'
/tmp/667876137/test.cpp:75:67:   required from here
/tmp/667876137/test.cpp:33:6: error: static assertion failed: FAIL
/tmp/667876137/test.cpp: In instantiation of 'class Fail<400>':
/tmp/667876137/test.cpp:78:28:   required from 'class expr_FLAPERON_THROW_MAX<400>'
/tmp/667876137/test.cpp:79:45:   required from here
/tmp/667876137/test.cpp:33:6: error: static assertion failed: FAIL
/tmp/667876137/test.cpp: In instantiation of 'class Fail<65535>':
/tmp/667876137/test.cpp:80:28:   required from 'class expr_UINT16_MAX<65535>'
/tmp/667876137/test.cpp:81:29:   required from here
/tmp/667876137/test.cpp:33:6: error: static assertion failed: FAIL
/tmp/667876137/test.cpp: In instantiation of 'class Fail<32767>':
/tmp/667876137/test.cpp:84:28:   required from 'class expr_INT16_MAX<32767>'
/tmp/667876137/test.cpp:85:27:   required from here
/tmp/667876137/test.cpp:33:6: error: static assertion failed: FAIL
/tmp/667876137/test.cpp: In instantiation of 'class Fail<18>':
[... 10 lines cut ...]
/tmp/667876137/test.cpp:33:6: error: static assertion failed: FAIL
/tmp/667876137/test.cpp: In instantiation of 'class Fail<4>':
/tmp/667876137/test.cpp:106:28:   required from 'class expr_ADC_CHN_MAX<4>'
/tmp/667876137/test.cpp:107:31:   required from here
/tmp/667876137/test.cpp:33:6: error: static assertion failed: FAIL
/tmp/667876137/test.cpp: In instantiation of 'class Fail<2250>':
/tmp/667876137/test.cpp:108:28:   required from 'class expr_PWM_PULSE_MAX<2250>'
/tmp/667876137/test.cpp:109:35:   required from here
/tmp/667876137/test.cpp:33:6: error: static assertion failed: FAIL
/tmp/667876137/test.cpp: In instantiation of 'class Fail<250>':
/tmp/667876137/test.cpp:114:28:   required from 'class expr_HEADING_HOLD_RATE_LIMIT_MAX<250>'
/tmp/667876137/test.cpp:115:63:   required from here
/tmp/667876137/test.cpp:33:6: error: static assertion failed: FAIL
In file included from /tmp/667876137/test.cpp:16:0:
./src/main/flight/servos.h:47:3: warning: 'inputSource_e' defined but not used [-Wunused-variable]
 } inputSource_e;
   ^~~~~~~~~~~~~

panic: no more matches

goroutine 1 [running]:
main.main()
	/home/jrh/Projects/fc/inav-fcgen/tools/settings.go:1120 +0x98
make: *** [Makefile:929: src/main/fc/settings_generated.h] Error 2
````

fiam, 2017-08-20:
@stronnag Just pushed a couple of commits which should fix both the issues with gmake 4 and GCC 7. Could you give it a try? Thanks!

fiam, 2017-08-20:
@stronnag Forgot to mention that the GCC 7 error is due to not having a reliable way to print an enum value at build time. I can get #defined'd constants with #pragma message, but it doesn't work for enums (since those are not handled by the preprocessor). If you can think of a better alternative, I'm all ears (or eyes?:-)).

stronnag, 2017-08-20:
It builds OK on make 4.2.1 / arm gcc 7.1 Good job!

fiam, 2017-08-20:
@digitalentity I'm not quite sure about what to do with the docs and there are also several issues with the toolchain installation on all the different platform/methods. At least we'd need to update the following docs:

* Building in Eclipse.md
* Building in Mac OS X.md
* Building in Ubuntu.md
* Building in Windows light.md
* Building in Windows.md

And the following toolchain support files:

* Dockerfile
* Vagrantfile
* install-toolchain.sh
* .travis.yml

And I'm probably forgetting something. I can do some of those myself, but for others I have no idea, specifically I'm totally clueless when it comes to Eclipse or Windows or Vagrant.

digitalentity, 2017-08-22:
@fiam update what you can - we'll fix others by combined effort

digitalentity, 2017-08-22:
@fiam it's probably time to open a PR with the code changes!

fiam, 2017-08-22:
@digitalentity Just finished rewriting the generator in Ruby, which should greatly simplify the updates that need to be done to the docs. I'll open a PR in the next few minutes.

fiam, 2017-08-22:
@digitalentity Submitted PR as #2028

stronnag, 2017-08-22:
Latest flown. built with ruby 2.4.1p111 and arm-gcc 7.1. No problems.

closed 2017-08-28 by digitalentity