iNavFlight/inav #2008: Erase blackbox dataflash chip via Stick Combination

giacomo892, 2017-08-18:
Since i've found that if the chip is full nothing is being recorded instead of overwriting old logs I think it could be very usefull if dataflash chip can be erased via a stick combination since I rarely connect my FC to a PC but having a BB records of last flight could be always crucial.

krzysztofmatula, 2017-08-18:
I'm with you, but I'm afraid there is no really convenient solution here...

Flash chip erasing takes long. ~1minute at least. Executing it manually before each take off would be a pain. An indicator how much space is free could help here.

I'm afraid the only robust solution is to switch to a FC with SD card slot.
Or use dirt-chip openlog device (which will consume one UART...)

giacomo892, 2017-08-18:
Erasing is not that slow. If the user initiate an erase via stick combination should expect to wait :)

digitalentity, 2017-08-20:
Hm... interesting option. However blackbox erase then should be made an arming blocker.

giacomo892, 2017-08-20:
Maybe we can even think about a cli setting that allow to mark as hardware health problem the fact that BB is full. Letting arming impossible

krzysztofmatula, 2017-08-20:
According to the N25Q064A datasheet (8MB flash chip used in SPF3) the "bulk erase" typically takes 60s, and that's what I observe in reality.

There's also a "sector erase" command that is said to typically take 0.7s (per 64kB long sector). This makes ~90s for erasing whole chip sector-by-sector, so having a background task preparing free space would allow (in theory) to have circular buffer in the flash, so the last flight log could be available without pre-erasing the chip in advance. This would also require a software buffer capable of holding ~0.7s of blackbox data that would be produced while the chip is blocked by a sector erase command. Not sure if this whole thing is worth spending time on programming though...

giacomo892, 2017-08-21:
@krzysztofmatula yes maybe the circular buffer approach is an overkill. I think we should make (optionally) arming non possible if dataflash is full and then let the user clean it via a stick command.

SoulPictures, 2017-08-21:
sounds a bit dangerous for me if we look at brownout behavior or FC reboot during flight.. if arming is not able due to "flash full" you could lose your aircraft due to this problem. maybe its better tu use something like the OSD to display it?

giacomo892, 2017-08-21:
@SoulPictures oh yes. Right!

Just stick to the original idea. Add the possibility to erase the flash via a stick combo (and/or switch) and then just prevent arming during the erasing process. One can easily erase the flash while waiting for a GPS fix and it is not really a issue.

krzysztofmatula, 2017-08-22:
I agree. The original idea is useful and relatively simple.

giacomo892, 2017-09-12:
I've found something similar is already in BF (only MODE not stick combination)
https://github.com/blckmn/betaflight/commit/ecaccebc021825050c4e8f7cd1e44d630c41d627

fiam, 2017-09-18:
@giacomo892 I'm not a fan of the stick combination. They're hard to remember and a bit "overloaded" (i.e. there are already too many to remember all of them). What about trigger by mode and by CMS? (the latter would require an OSD)

giacomo892, 2017-09-18:
Yes it is ok. also would be nice to have like BB full overlay :)

closed 2018-05-27 by stale[bot]

marianomd, 2023-04-14:
This should be refloated.