iNavFlight/inav #2010: RTH throttle issues..

cam193, 2017-08-19:
I went out for a flight today as normal, with a brand new SP racing f3 flight control and I went up to around 300 feet and about 1000 feet away from me just to test out the RTH because of some issues I had been having, hence the replacement of the flight control. As I turn on RTH it turns towards me, flies for a few feet and then full throttles, drops, full throttles, drops, each time lasting about 5 seconds per full throttle / drop, with absolute no control on my part. As this was happening I was freaking out and I turned RTH off and back to angle mode and that did not do anything. I disarmed the drone, still nothing, still doing the full throttle thing. Finally after about a minute of this it finally just fell out of the sky and went pow as it slammed the ground. Still flies! But I am wondering what is causing this issue and wondering if it is a known issue. Thanks. I can post pictures of the fight controller location of need be. I have it high off the frame to eliminate interference. Any help would be appreciated! :)

digitalentity, 2017-08-19:
If it exhibits this behavior even in ANGLE mode and even when you attempt to DISARM - I see only a few possible reasons:
 1. Faulty hardware or wiring (FC or ESC)
 2. Misconfigured failsafe **on receiver side**

Please provide as many details as possible - blackbox log would be very nice to have.

cam193, 2017-08-19:
I have failsafe set to RTH in both the receiver and the FC. And I'm pretty sure the wiring is correct- It works just fine in angle but only freaks out when I use RTH. I am not sure if the blackbox log logged everything. It was brand new so I am not sure if it logged it all. I never cleared it since I got it but tomorrow I can wipe it and try to log the event again. Unless I am wrong and it logs it no matter what?

cam193, 2017-08-19:
The equipment I am using is SP Racing f3 deluxe as FC, Neo m7n as GPS from banggood, Futaba 10J as TX, Futaba RS3008SB as RX, Using SBus to the FC, Polaris 35 amp ESC's

ymuraki1, 2017-08-29:
I have very similar issue.
All angle mode and air mode fly perfect, but as soon as I set to position hold or RTH, model takes off vertically.  I aborted RTH/PosHold bring model back, but it's repeatable.
I played with PID for baro, and GPS Vertical, but seems to have no effect.

From looking at blackbox log, as soon as I hit RTH or PosHold, throttle goes to 100%. (Timestamp around 20S.)  
I checked my radio input for the mode (w/o battery connected) and nothing is commanding 100% throttle from radio.
My equipment is OmnibusF4V3 with 250mm Quad, Futaba R2001SB (Sbus) receiver.
I had very same issue with SPRF3 with different GPS.  It was working just fine with V.1.6, but as soon as I updated firmware to1.7, quad started jump high on RTH.  I thought I killed my FC, so I changed to OMNIBUS.
So problem maybe at firmwares or configurator.

[LOG00006.TXT](https://github.com/iNavFlight/inav/files/1260349/LOG00006.TXT)

digitalentity, 2017-08-29:
@ymuraki1 you have a well known vibration/resonance issue. Once you go above certain throttle (in your case ~45%) accelerometer is reading garbage. Since navigation is highly dependent on accelerometer - it's messed up. Soft-mount your FC and/or your motors.

ymuraki1, 2017-08-29:
Thanks digitalentity,  I'll softmount FC.
But my question is  why airmode or angle mode works without any issue?
Also, the navigation mode make it throttle to 100% uncommanded that quickly?

digitalentity, 2017-08-29:
Acro and Angle mostly relies on gyro which is fine with that level of vibration. 

When accelerometer goes crazy it's reading zero *on average*. What's zero acceleration? It's free fall. So firmware (altitude hold code) thinks that machine is falling and kicks up throttle in attempt to compensate for this free fall (as seen by sensors).

ymuraki1, 2017-08-29:
That makes sense.

ymuraki1, 2017-09-02:
I iosmounted FC.
Problem fixed. 
I put the PID parameter for GPS and Baro to default and seems nav flight is working fine.

Thanks for your help!

closed 2017-09-03 by digitalentity