iNavFlight/inav #2011: LTM telemetry on SS issue

VictorBucha, 2017-08-19:
LTM telemetry via Bluetooth. EZ-Gui sometimes reports "no data received". Symptoms similar to  those in #1990. Tried changing I2C freq at 800, 400,200 KHZ with no effect. Telemetry via USB works flawless. FC- SpracingF3 Acro. Tried both 9600 and 19200 baud with no effect.

teckel12, 2017-08-21:
@VictorBucha does it only happen when disarmed?  When armed, telemetry is never dropped, it's only when disarmed when it happens (for me at least using SmartPort over soft serial).  I've had a few long-range telemetry drops, but that's expected.  I can arm and burn through an entire battery (just sitting on the ground) and telemetry will never drop.

VictorBucha, 2017-08-21:
Both armed and disarmed. In a few cm from BT dongle. When plugged via usb- works perfectly

giacomo892, 2017-08-21:
I have a similar issue using software serials and HC12 at 9600baud.
After flying with LTM enabled , disarming, ezGUI switches again to MSP as expected and then I cannot arm the craft anymore and telemetry stops to answer.

teckel12, 2017-08-22:
I've seen heat cause a problem.  I can literally blow on my quadcopter and telemetry will start again.  Which is why it never drops when armed, because the blades are cooling the quadcopter.  If I turn off my VTx the problem also goes away as the VTx generates more heat than most other components.

giacomo892, 2017-08-22:
I'm using a custom board and a relativity big quad  with plenty of space without heating issues. I don't think that heat is the problem here.

teckel12, 2017-08-23:
@giacomo892 With the current release of iNav there's a timeout that will kill the software serial port.  My pull request #2002 has changes that remove that timeout (it doesn't work as intended anyway and it's been removed from Cleanflight/Betaflight already).  Not sure if my changes are going to be part of iNav v1.8 or not, but when #2002 is merged try this again to see if it corrects your problem.

digitalentity, 2017-08-23:
#2002 doesn't seem to touch softserial code, what timeout are you talking about @teckel12?

VictorBucha, 2017-08-25:
If connected when disarmed, it constantly repeats "no data received".  When connected armed, there are only 3 to 5 seconds pauses and if disarmed- data flow goes the same. But pauses don't disappear at all.

giacomo892, 2017-08-25:
If you switch to CLI and go back to the program does everything restores and works?

VictorBucha, 2017-08-25:
Not able to test. I had crashed my heart today. Was testing Failsafe. Drone was remaining in the alt+GPS poshold mode and not landing. So it fell down while I was rearming the Transmitter.

teckel12, 2017-08-25:
@digitalentity You're right, #2002 would be specific to SmartPort telemetry, so it wouldn't apply to LTM.  Maybe the same timeout is in LTM telemetry?  I didn't mess with LTM as I don't have that telemetry.

giacomo892, 2017-08-25:
@victorbucha so sorry  I hope you have logs. Were you under or over the RTH altitude?

VictorBucha, 2017-08-25:
@giacomo82, unfortunately, I don't. I switched off logging to reduce cpu load

closed 2018-05-27 by stale[bot]